FD-/RPiPlay #96: iOS 13.4.1 problems

sorriso93, 2020-04-13:
Hello I just discovered what it seems an incompatibility with iOS 13.4.1 and iPadOS 13.4.1. Before the upgrade was all fine, now with my iphone and ipad I get this and a blank screen when I try mirroring

Accepted IPv4 client on socket 25
Local: 192.168.0.18
Remote: 192.168.0.70
raop_ntp receive timeout

Please note iphone of my wife mirrors with no problem and also my mac with 10.15.4 gets mirror with no problem

FD-, 2020-04-13:
Working fine for me on iPadOS 13.4.1.

sorriso93, 2020-04-13:
really weird, 3 iphone, 1 ipad pro and a mac. only on devices (1 iphone and 1 ipad) with 13.4.1 is not ok for me... But three days ago was ok on all devices at least at the first mirroring, I will investigate again...

sorriso93, 2020-04-13:
A simple reboot of the iphone resolved... no komment

closed 2020-04-13 by sorriso93

CHANist, 2020-05-02:
I have the same problem on using it in iPad OS 13.4.1. Just want to ask @sorriso93, is your problem solved for your ipad pro on 13.4.1?

sorriso93, 2020-05-02:
No more problems after i rebooted the devices