FD-/RPiPlay #97: Feature request: script hooks

jiridurdis, 2020-04-14:
Hi,

Would you mind to add "pre-start" (after connection, before starting video rendering) and "post-stop" (after disconnect) script hooks feature?

Examples of what could be done with hooks:
- change output resolution and let the TV handle the upscaling (and change it back after disconnect)
- terminate running apps that could affect video performance, network bandwidth etc.

From my point of view it should be sufficient to handle one "pre-start" and one "post-stop" script argument (both optional).

In "pre-start" phase it might be useful to watch and enforce fast script execution time (terminate hook execution after for example 3-5 seconds).

P.S.: Thanks for your work, RPiPlay is really good.

pallas, 2020-04-30:
Please see the script I posed to Issue #82.

closed 2020-10-06 by pallas

DaCHack, 2023-03-05:
@pallas 
I am still trying to find out how to use RPiPlay in parallel to Kodi on LibreElec.
As the system freezes everytime a Connection is established and screen mirroring is about to start, I assume I will need a hook before and after the connection to close Kodi and restart again.

In #131 you were recommending the script due to race conditions. Do they still apply? What hinders us to just run a predefined external script somewhere after line 70 in lib/raop.c? Sorry if the question seems naive but my c++ skills are close to zero 🙈