FD-/RPiPlay #98: FYI. Unix version of AirPlay server.

antimof, 2020-04-15:
Universal unix version of this project based on Gstreamer. Early prototypt but fully working.
https://github.com/antimof/UxPlay

FD-, 2020-04-15:
Fantastic! Are you planning to upstream the new renderers once you have them working? Does anything besides the renderers need modifications?

antimof, 2020-04-15:
Sure I will do it

FD-, 2020-04-15:
Cool! Let me know if you have any questions regarding implementation of the current renderers!

antimof, 2020-04-15:
No. It works on any UX platform. The only concern is accelerated video sink. It even works on Raspberry pretty well with custom Gstreamer plugin gst-mmal for accelerated decode and screen output.

FD-, 2020-04-15:
I'm particularly looking forward to seeing how the gstreamer renderers perform on the Pi Zero. Does gst-mmal support audio via USB audio cards or is audio handled by a different plugin anyway?

antimof, 2020-04-15:
gst-mmal uses OpenMAX hdmi output. However, audio decoder is software only....

FD-, 2020-04-15:
Audio decoding is software-only in the current renderers as well (AFAICT, the Pi doesn't have any hw audio decoders), so I'm hoping to see similar performance + possibility to use USB audio interfaces.

pallas, 2020-04-30:
Should this work be wrapped back into RPiPlay? Definitely being able to build x86-64 will help my current debugging efforts. At which hash did you fork the code?

antimof, 2020-05-07:
I think it was 5b2193c68e9b2c8f807dc56980b885cc12fd6043.

pallas, 2020-10-06:
This was merged.

closed 2020-10-06 by pallas