FD-/RPiPlay #100: [HELP] It doesn't works

Albertosogasol, 2020-04-24:
I have followed all the steps but when i inicialize the executable i cant mirror my iPhone on the screen.
The RPi appears on my iPhone and I can connect but nothing else happens. I tried with different iPhones but I have the same issue.
This is the log in the console:

**~/Desktop/RPiPlay/build $ ./rpiplay**

Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.1.34
Remote: 192.168.1.35
Accepted IPv4 client on socket 17
Local: 192.168.1.34
Remote: 192.168.1.35
raop_rtp_mirror starting mirroring
raop_ntp receive timeout
raop_rtp starting audio
^CStopping...

Raspbian is up to date.

Does anyone has the same issue? 

Thank you!

oldsadsongs, 2020-04-24:
Same issue -

Albertosogasol, 2020-04-24:
Ok I solved it.
VNC was the the problem. I was inicializating it through VNC, but I tried to open the executable out of VNC and it works, but there is antother problem. The app freezes and i cant kill it. I have to open the terminal from my PC via SSH and kill the process, because the mouse and keyboard connected to the Raspberry doesnt work

FD-, 2020-04-24:
Well, if this turns out to have been the problem for everyone reporting similar issues, that’d be fantastic. And it makes perfect sense, I just didn’t think people would by default assume RPiPlay to work via VNC. Since it uses the GPU for decoding the H264 video data, RPiPlay’s output doesn’t go through the frame buffer that VNC captures. However, it is possible to use RPiPlay via VNC, you just have to enable an extra setting in the VNC server: See [this](https://www.raspberrypi.org/blog/minecraft-pi-and-more-over-vnc/) or [that](https://www.raspberrypi.org/documentation/remote-access/vnc/).

closed 2020-04-24 by FD-