FD-/RPiPlay #101: [Question] Capture Cast

reteps, 2020-04-25:
I would like to take a screenshot of the screen being mirrored (ideally a video stream would work). The end goal is to process the screenshot and if I find certain text on the screen I run a script. However, when I run the raspberry pi screenshot utility, it appears to *screenshot the area under the mirror*!

My command to run the script is:
` ./rpiplay -n "Raspberry Pi" -b off -a off`

And to screenshot:
`scrot -a 750,450,400,400 image.png`

Why is this happening?

minoskt, 2020-04-25:
Same issue is referred here #100 . The script uses the GPU to decode the video. You can fix it by enabling the Direct Capture Mode in VNC Server.

reteps, 2020-04-27:
I would like to process the stream with a FFMPEG script for every frame. Is this possible now that I am using a VNC server with it?

pallas, 2020-05-10:
If you set `DUMP_H264`, you can write the video stream as it is being mirrored. YMMV

mike5im5, 2020-05-11:
I had a similar problem capturing using scrot. The solution for the other issue was specifically around VNC (which shared the overlay issue).

I managed to use this solution to capture the entire overlay from command line:

https://www.jonwitts.co.uk/archives/728

closed 2020-06-07 by FD-