FD-/RPiPlay #111: Screen resolution...

gioboske, 2020-05-02:
Hi,
I've installed RPiPlay on a Raspberry Pi ZeroW and everything works out-of-the-box!

The only issue I'm experiencing is that the screen resolution when using the RPiPlay as extended monitor is not correct and it is not the same as the RPi default resolution (black bars on top and bottom of the screen).

Is there any option to set the required resolution on RPiPlay?

Thanks for your support.

pallas, 2020-05-04:
Have you attempted to tune the overscan on the rpi or turned off overscan on the monitor?

gioboske, 2020-05-09:
> Have you attempted to tune the overscan on the rpi or turned off overscan on the monitor?

Yes. I tested different configuration on monitor side (also disabling overscan).
Also I've tried to tune the overscan on the raspberry.

Anyway, I think this is not the problem, since when I switch on the raspberry everything works correctly (no black bar at all). I experience the issue only when RPiPlay starts and the remote desktop is shown.

Thus, I guess the issue is not resolvable playing with general raspi or monitor settings, but changing something on the RPiPlay side, instead.

What do you think?

FD-, 2020-05-09:
This sounds a lot like you forgot to set the display resolution for the AirPlay display on the macOS side. Go into macOS system settings -> Display and set the RPiPlay screen's resolution to 'Default for display". For whatever reason, it seemed to default to 480p when I tried it a few seconds ago.

gioboske, 2020-05-09:
Unfortunately the issue is not solved changing the resolution on macOS side.
I've tried 'Default for display', 1080p and 720p, but the top and bottom black bars persist.
It is quite a strange behaviour. Really I do not get which is the problem...

FD-, 2020-05-09:
What resolution does your monitor have?

gioboske, 2020-05-09:
> What resolution does your monitor have? Is it probably not 16:9?

It is a 16:10 aspect ratio monitor, 1680 x 1050@60Hz.
Anyway, I use RPiPlay as extended monitor from an macbook air (late 2015) with a screen resolution of 1440x900, so with the same aspect ratio.

FD-, 2020-05-09:
RPiPlay's resolution is hardcoded to 1920x1080 (https://github.com/FD-/RPiPlay/blob/6514932aab7e2d60e9d9170a0cbe0c1eb8f19e77/lib/raop_handlers.h#L137). I suspect that's why you're seeing bars on a different aspect ratio.

gioboske, 2020-05-09:
Ok, thank you. I agree with you, that is probably the source of my problem.
Is there any plan to support different resolutions/aspects ratio in the near future?
If not, I guess I can try to change the hardcoded resolution and recompile RPiPlay. What do you think?

FD-, 2020-05-09:
Yeah, please try changing the hardcoded values and let me know if that solves your problem.

gioboske, 2020-05-10:
> Yeah, please try changing the hardcoded values and let me know if that solves your problem.

Recompiled after changing the hardcoded values and works like a charm.
Now from the display preferences in macOS I can set the correct resolution (1680x1050). No top or bottom black bar anymore.

I think that it could be a good improvement to configure the resolution directly from the command line. Anyway, recompile the code does not take so long and I plan to use RPiPlay on a single monitor. So not a big issue.

Thank you for your support!

closed 2020-05-10 by gioboske

FD-, 2020-05-10:
Ideally, we'd query the actual monitor resolution from the Pi, probably by getting dimensions from /dev/fb0. Shouldn't actually be too difficult, just don't have time to look into it at the moment.