FD-/RPiPlay #112: Only running audio / crashes from Apple Music

Levilutz, 2020-05-05:
Trying to use this to just stream audio, but it appears there's no option to disable screen mirroring and just accept audio. 

Potentially part of the same issue, attempting to connect to the device from Apple Music on a computer just gives the error message in Music: 
`The AirPlay device “RPiPlay” cannot be found on your network. Please verify your network settings and AirPlay configuration or select another device.`

Nothing shows up from RPiPlay when attempting to connect, even with the debug option on. Both devices are on the same network, and it doesn't appear anything is blocking the connection. 

Intrestingly, screen mirroring works fine so long as I don't play music. However, if I use Apple Music over AirPlay while mirroring, the mirror ends and my raspberry pi appears to freeze. I can still ssh into the pi and kill the rpiplay process, but the screen stays stuck in the last frame of mirroring until I restart. 

Trying to get debug logs of what gets printed when it crashes, so I'll put them here when I can.

Levilutz, 2020-05-05:
[testlog.log](https://github.com/FD-/RPiPlay/files/4578776/testlog.log)

Seems like it crashed halfway through printing "Got h264 data of XXX bytes". Printing wasn't cutoff from killing the process; that was the last log message before it froze. 

Interestingly, audio did stream for a bit this time. Pausing the music and playing again, however, caused it to crash.

pallas, 2020-10-06:
Adding this to my TODO list.

FD-, 2020-10-06:
Regarding connecting directly from Apple Music, my guess is that tries to use the separate audio-only protocol (ALAC audio), while RPiPlay only supports the mirroring protocol (AAC audio).