FD-/RPiPlay #113: Rotation issue

sdmydbr, 2020-05-05:
Well I don’t know if I’m the first one Or it’s related only with mine, but when I rotate the screen, the screen keeps on flickering. Works fine without rotation.

FD-, 2020-05-05:
Please include as much useful information as possible.

sdmydbr, 2020-05-05:
Alright so I was using it without any screen rotation and it works just fine but when I rotate the screen by ‘n’ degrees the screen flickers..

FD-, 2020-05-05:
What Pi? What OS? What iOS version? What iOS device? Come on...

sdmydbr, 2020-05-05:
Well running rasbian on raspberry pi4 2 gb variant, iPhone running iOS 13.4.1

FD-, 2020-05-05:
I don't have a Pi 4 to test with. Problem could be related to #73

sdmydbr, 2020-05-05:
Maybe related to pi4 then..

FD-, 2020-05-05:
Do you probably have an older Pi to test the same SD card image with?

sdmydbr, 2020-05-05:
> What Pi? What OS? What iOS version? What iOS device? Come on...
>
> <https://github.com/FD-/RPiPlay/issues/113#issuecomment-624135334>, or
> unsubscribe
raspberry pi 4 2gb variant , on raspbian, IPhone 7 running iOS 13.4.1

FD-, 2020-06-07:
Did the recent changes to rotation fix the problem?

sdmydbr, 2020-06-18:
Yes it is fixed thank you.. works like a charm🤩

closed 2020-06-18 by FD-