FD-/RPiPlay #115: Black Screen and can’t stop

lionelbachelart, 2020-05-07:
Hello there,
i did set up the program as you described in the ReadMe. When i am starting the program, i’m having the problem, that there is only the Black Screen. Tried it on my iPhone and iPad (latest iOS version) but it doesn’t work.
Also i can’t stop the program. Help please.

Thank you :)

FD-, 2020-06-07:
Did PR #116 fix the issue? Please note a black screen is to be expected when starting RPiPlay. How did you try to connect from your iOS devices? Did you use the AirPlay mirroring control as opposed to the AirPlay audio control?

adotagarwal, 2020-08-20:
I have this same issue, running on an RPi 4. I start the process and it blanks the screen (unless i choose -b auto) and then the process becomes unresponsive. Any attempt to kill it just results in the process showing up in brackets when you run ps. Even a reboot fails, the only way to recover is to remove power.

adotagarwal, 2020-08-20:
I figured out this was because I am using the DSI display. By setting HDMI resolution in raspi-config, the program now starts normally. Is there a way to choose which display to use for video output?

adotagarwal, 2020-08-20:
I managed to modify the source code to use displays 2, 4, or 7 for the background and video render,

pallas, 2020-08-20:
Could you please make a pull request of the patch?

closed 2020-12-19 by pallas