FD-/RPiPlay #118: FYI: fuzzing h264bitstream, loads of issues

pallas, 2020-05-09:
I decided to fuzz this dependency and even though it is only used to inject frames that lower the pipeline delay there are a huge number of streams that cause it to crash. I'm looking to see if there is a reasonable way for me to address them upstream.

This is just an FYI re: that.

pallas, 2020-05-10:
See https://github.com/aizvorski/h264bitstream/issues/36

pallas, 2020-05-16:
So I've locally disabled `max_dec_frame_buffering` on my build, as this is a huge security risk. There are buffer overruns and double frees all over the place.. basically every array access is suspect. Although I do not have a POC it's hard to imagine that this is not exploitable.

I submitted a PR upstream but haven't heard anything since then, and the fuzzer found more issues over the week.

I just cloned VideoLAN's x264 (GPL 2.0) and I'm going to see if it would be safer & easier to use that. The upside is that it's available via apt.

pallas, 2020-05-16:
Oh, except it doesn't have a decoder :(

FD-, 2020-05-24:
Yeah, I think the h264bitstream library was the only one that offered what we need, out of many GitHub projects a search for 'H264' yielded back when I implemented this.

pallas, 2020-05-24:
That sounds right to me. I also looked for a replacement and AFAICT there isn't one. :(

pallas, 2020-10-06:
I'm still of the opinion that what this library is doing is quite dangerous. Would it be worth disabling in the default build?

FD-, 2020-10-06:
In my tests back when I implemented that part, audio was noticeably off when it's disabled, but I guess I could live with adding a -s(ync) option instead of the -l option and disabling it by default. Would you prefer to exclude the library in the default build altogether?

pallas, 2020-10-06:
That's a good point.. if we're doing low-latency, the audio will already be out of sync right? So maybe it's OK to skip the bitstream stuff if we're in low-latency mode??

The problem with the library is that it's easy to cause a buffer overrun just by sending bad h264 data. It skips all kinds of bounds checking, &c, and in my build I do disable it unconditionally. I tried to fix the upstream problems but there were just a ton.

pallas, 2020-12-19:
The upstream project doesn't seem interested in not crashing. Closing for now.

closed 2020-12-19 by pallas