FD-/RPiPlay #121: FDKlibInfo_lookup, FDKlibInfo_getCapabilities, FDKinitLibInfo

ldon13, 2020-05-17:
Followed the Installation, Using Latest Raspbian
```
Scanning dependencies of target renderers
[ 98%] Building C object renderers/CMakeFiles/renderers.dir/audio_renderer_rpi.c.o
In file included from /home/pi/RPiPlay/renderers/fdk-aac/libAACdec/include/aacdecoder_lib.h:457,
                 from /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:33:
/home/pi/RPiPlay/renderers/fdk-aac/libSYS/include/FDK_audio.h:757:29: warning: ‘FDKlibInfo_lookup’ defined but not used [-Wunused-function]
 static FDK_AUDIO_INLINE INT FDKlibInfo_lookup(const LIB_INFO* info,
                             ^~~~~~~~~~~~~~~~~
/home/pi/RPiPlay/renderers/fdk-aac/libSYS/include/FDK_audio.h:745:1: warning: ‘FDKlibInfo_getCapabilities’ defined but not used [-Wunused-function]
 FDKlibInfo_getCapabilities(const LIB_INFO* info, FDK_MODULE_ID module_id) {
 ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/RPiPlay/renderers/fdk-aac/libSYS/include/FDK_audio.h:735:30: warning: ‘FDKinitLibInfo’ defined but not used [-Wunused-function]
 static FDK_AUDIO_INLINE void FDKinitLibInfo(LIB_INFO* info) {
```

ldon13, 2020-05-17:
Ignore this issue, I launched the execution and it works. Thanks for this!

ldon13, 2020-05-17:
How do exit the application after disconnecting from iOS? Getting a Black screen.

pallas, 2020-05-24:
Those are harmless warnings based on the way fdk-aac is being linked in. There are tons of other threads here about the second issue you are reporting. Please take a look at them, e.g. Issue #86.

closed 2020-05-24 by FD-