FD-/RPiPlay #122: Failed to execute child process "xterm" (No such file or directory)

PWKPR, 2020-05-23:
I installed but when I execute the executable, I just get a blank screen. My iPhone 7 and iPhone 6+ can see RPiPlay, however I just get a black screen with no audio or video, and I can't get to do anything but restart the Pi from the powersocket. (First I thought it was crashing the Raspberry Pi but since the iPhones can see the Raspberry Pi and RPiPlay in Airplay, I think it's just a black screen.)

However, when I execute in terminal I receive the error: "Failed to execute child process "xterm" (No such file or director)

This is my first Raspberry Pi project, so looking forward to getting this going.

pallas, 2020-05-24:
That error is unrelated to this project.

FD-, 2020-05-24:
To add to what @pallas wrote: The error message you're seeing ('Failed to execute child process xterm') is unrelated to RPiPlay. Regarding the black screen, I suggest you search through the list of issues other users had with RPiPlay. I think some users solved a similar problem by increasing the GPU memory split.

closed 2020-05-25 by FD-

PWKPR, 2020-06-07:
Amazing. Just fixed it now. 

Thanks so much for your help. This is awesome 

Brian

Sent whilst on the move: typing emails, dodging lampposts.

www.ParkerWayneKent.com