FD-/RPiPlay #124: build problem with fdk-aac

alxt17, 2020-05-25:
After the command "cmake ..", the terminal shows **"You have called ADD_LIBRARY for library fdk-aac without any source files. This typically indicates a problem with your CMakeLists.txt file".**

However, I could still build it since it shows "Configuring done" and "Generating done". But when I run the program and use the airplay on my iPhone, the terminal on raspberry pi constantly showing **"aacDecoder_DecodeFrame error"** which I assume there is something to do with the warning about fdk-aac as the above.

How can I fix that? Thank you!

FD-, 2020-05-25:
Are you sure this is unrelated to #2 ? Make sure you're connecting via the Mirroring control.

alxt17, 2020-05-25:
> Are you sure this is unrelated to #2 ? Make sure you're connecting via the Mirroring control.

Sorry that I overlooked that issue and appreciate for replying so quickly!

closed 2020-05-25 by alxt17