FD-/RPiPlay #128: Audio in non-low-latency mode instable

DaCapitalist, 2020-05-30:
Hi FD- and all,

first of all let me thank you for such a nice and helpful peace of code making lifes really easier!
I am not a really experiences programmer, but I made my first steps into several markup- and script languages. So I hope eventually to be able to support also this project.

Now my issue: When using low latency mode, audio and video streaming in mirroring mode seems to be quite stable yet maybe sometimes a little choppy (as you point out correctly).
When I run in normal mode, the first video or song to play via apps such as Spotify, DSfile or Safari (youtube) usually works well, but after a short time of skipping songs or just closing a video being played and continuing later on results in either
- audio being replaced by an extreme grumble (you can hear that it should be the song, but it is completely decomposed)
- no audio at all anymore
Video continues to work well in any case and also the console output looks quite normal (see example below)

Anyone else experiencing this issue? Any idea how to solve it or where in the code the issue could arise?

`Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.1.11
Remote: 192.168.1.20
Accepted IPv4 client on socket 17
Local: 192.168.1.11
Remote: 192.168.1.20
Found an unknown parameter: volume
raop_rtp_mirror starting mirroring
raop_ntp receive timeout
raop_rtp starting audio
raop_rtp starting audio
raop_rtp starting audio
raop_rtp_mirror tcp socket closed
Connection closed for socket 15
Destroying connection
Connection closed for socket 17
Destroying connection
^CStopping...
`

FD-, 2020-06-07:
Did the recent improvements (commit fa238231c8b7c26aa77c9e938661795a6ebbd500) by @pallas improve the situation?

DaCapitalist, 2020-06-07:
My RP4B moved to another position so I will need to test with a new RP Zero W yet to be purchased.
Will do after I find one to a regular price. Seems like stocks are surprisingly low right now :(

dgthomson, 2020-06-13:
I installed RPiPlay on a 3B yesterday and no, the audio issues described by dacapitalist remain.

The pattern I have noticed on Youtube in particular is identical. Solid video, transient audio until a new Youtube video starts or an advert plays, then no audio or noise (it sounds like an old Commodore 64 loading from a tape!)

Console messages are the same-- if I change the volume on the mac's keyboard, I get additional 'raop_rtp starting audio' messages for each button push.

I agree with dacapitalist- this package has come on a lot since the last time I looked and I will continue using it as I won't need sound, many thanks.

Happy to fiddle to find a cause if I have instructions...

DaCapitalist, 2020-06-13:
My RPI Zero WH is finally on its way. so from Wednesday next week I will probably be able to fiddle around again.

Just needed to Google what a C64 Sounds like loading from a tape :). but i do not remember the sound to be that bad. More into the direction of an old analog TV without signal. But could be driven by the speakers.

My Setup: the RPI is connected via HDMI to a  Speaka HDMI to 6CH Audio Decoder. From there the converted analog Signal goes to a Teufel 5.1 set. Works like a charm with miracast cable adapters connected to the iPhone but as described not yet seamlessly from the RPI

pallas, 2020-06-13:
Are you running in low latency mode or synchronized mode? I don't quite trust the sync version of the pipeline and always run in low latency mode (-l on command line) because of that. Also see my upstream branch with future fdk merges.

~Derrick • iPhone

DaCapitalist, 2020-06-16:
@pallas As described in my First post I primarily use the low latency mode due to the described issues. It does its job but you notice that it can sometimes be a bit choppy.
Will try and test it out a bit more beginning Thursday...

Ijdtm7, 2020-06-25:
I am also having extreme audio issues on my pi 3 B+ if I’m not using low latency parameter

flatsiedatsie, 2020-07-01:
Audio was a glorious and fascinating garbled mess until I used the `-l` option.

dgthomson, 2020-07-09:
Confirm the above, starting with '-l' fixes the audio. It's lifechanging to Airplay a Teams meeting to the big TV which I can participate in while lying on the couch.

flatsiedatsie, 2020-07-10:
As a sidenote: I've intgrated RPiPlay in an addon for the [Mozilla WebThings Gateway](https://iot.mozilla.org/gateway/). Users of this open source smart home system can now install RPiPlay with one click, and manage it through a simple GUI.

jotoid, 2020-09-25:
No audio over HDMI on RPi4 after I downloaded latest version.  Throws a "Found an unknown parameter: volume". Still having #71 issue with analog so casting is unusable for my use cases.

pallas, 2020-10-06:
This thread is hard to follow for me but my take-away is that low-latency mode fixes the issue. I'm going to go ahead and close. Unfortunately, fixing sync is outside of my timebox at the moment. I'm hoping that the wonderful gstreamer work by @antimof and @dougg3 will eventually enable real OpenMAX audio/video syncing in gstreamer itself. Until then, I'd recommend turning off sync (i.e. use `-l` for low-latency) and disabling the dangerous h264-bitstream twiddling.

closed 2020-10-06 by pallas