FD-/RPiPlay #129: Unable to detect RPiPlay Server

ArcherN9, 2020-05-31:
Hi, 
Thanks for maintaining this project. From the looks of it, it's not an easy feat!

All of my devices fail to detect RPiPlay on the home network.
1. iPhone 7+: 13.3.1
2. iPad Air 2: 13.3.1
3. iPhone 11: 13.3.1
4. MacBook Pro Mojave 10.14.6

I'm running RPiPlay on a Pi4 2GB with DietPi. I've followed all the steps on ReadMe and the issue list and was able to get it installed so that when I run it, I get the following:

```
dietpi@Pi-4-Hub:~/Downloads/RPiPlay/build$ sudo ./rpiplay -n "PiHub" -b auto -a hdmi -d
> stream info: channel = 2	sample_rate = 44100	frame_size = 480	aot = 39	bitrate = 0
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
```

My `config.txt` under `/boot` is:
```
#-------GPU memory splits-------
gpu_mem=128
gpu_mem_256=128
gpu_mem_512=128
gpu_mem_1024=128
```

What I've tried so far
1. I followed #10 and got all moved to `/opt/vc`
2. Followed #8 and increased `gpu_mem` to 128
3. Followed #91 and modified blanket policy of INPUT to ACCEPT
4. Followed #64 - Rebooted RPI and all of my iOS devices numerous times, installed RPiPlay twice.

Other information:
1. The black screen shows up on the TV connected via HDMI when I run the executable with the flag `-b on`
2. Pi4 is connected to my home router via eth0 and other devices, are obviously on WiFi. 
3. Other services on the Pi - PiVPN, Transmission, Plex work fine.

I understand you do not extend explicit support beyond Raspbian. But anything indicative as well would be greatly appreciated.

DaCapitalist, 2020-05-31:
Hi Archer,
Of you See the black Screen the issue seems to be on the network.
Did you try with transmission and vpn services disabled? Can you reach RPiPlay‘s ports via nmap from a computer in the network? Your router could block packets or they could be routed into VPN due to your VPN configuration.

ArcherN9, 2020-05-31:
@DaCapitalist Good theory. I didn't think of it this way. Found the following:

1. This is a screengrab from my `iptables` INPUT filter. I've allowed the ports described [here](https://nto.github.io/AirPlay.html), [here](https://support.apple.com/en-us/HT202944) & [here](https://support.airserver.com/support/solutions/articles/43000512423-what-ports-bonjour-services-are-used-by-airserver-universal-can-i-override-them-)
[image]

2. However, when I use nmap to scan these ports, they show up as closed which is a bit weird. I would have accepted nmap's response in this case but everytime I trigger the scan, the `pkts` and `bytes` counter against the `iptables` entries on RPi increases leading me to believe RPI accepts packets properly on those ports.

[image]

3. To cross verify the response I receive from other ports that are defined open the same way, I executed another nmap scan. All ports defined below are tagged as open and a service is running against it.

[image]

Still the same, cannot detect any device when I tap on "Screen Mirroring" on the iPhones.

DaCapitalist, 2020-05-31:
Does it work if you fully disable the firewall?
This comment might kick-in in your case:

The way it's currently designed you'll need to open everything because RPiPlay lets the operating system pick random local ports.  If you were to hardcode the local ports you'd still need to open up everything outbound to the iDevice for audio synchronization to work.

_Originally posted by @inferiorhumanorgans in https://github.com/FD-/RPiPlay/issues/91#issuecomment-631812466_

ArcherN9, 2020-06-01:
Blanket ACCEPT policy was the first thing I tried. Even with the individual port entries, as shown above, the blanket ACCEPT policy remains in force. The entries were to test if any data is being accepted on any of the ports as specified by the specification. I did not notice any increase in the number of pkts against the ports on iptables.

FD-, 2020-06-01:
I suggest you set up an isolated network (eg by setting up the Pi as a Wifi access point) just for testing.

DaCapitalist, 2020-06-01:
Just to ensure again: the VPN service is offline so that we are not talking about a routing issue or something around RPiPlay opening sockets on the wrong interface?

ArcherN9, 2020-06-03:
As suggested by @FD-, I did a fresh installation of DietPi on a spare RPI Zero. Since it was just for testing, no additional service was configured on it besides `proftpd` for sharing `vc` files to `/opt/vc`.

Additional info:

1. Fresh installation has ACCEPT all policy.
2. Since an RPiZero cannot be used as a hotspot for an isolated network, I used a spare DLink router instead. Not connected to the internet, just open WiFi for LAN. Enabled upnp on the router for good measure.
3. Just 2 devices connected the network - an iPhone and the Pi.
4. restarted all devices multiple times over the course of debugging.
5. Bumped `gpu_mem` to 256.

The issue persists, I don't see the RPiPlay name in any device.

@DaCapitalist Yes, the VPN service was offline on the Pi4 as of my last comment. The latest attempt on a PiZero is a fresh DietPi installation.

ArcherN9, 2020-06-09:
Thoughts, anyone?

DaCapitalist, 2020-06-09:
Did you try with a Raspbian lite to sort out any OS related issues?
Other than that I am pretty much puzzled.

I would try to turn off the memory Settings completely in the Boot config as a last idea and check if something changes.

ArcherN9, 2020-06-09:
I was thinking I should do that. I haven't so far. I wanted this to run with DietPi since that's what my Pi4 runs on. I tried reching out to Dietpi community on reddit and twitter. No one seems to bite. I may have to go to their forums :-/

Anyway, I'll try with Rasbian lite. It may weed out router/network related issues.

ArcherN9, 2020-06-13:
Tried it on PiZero with RaspiLite. Works as intended but without the sound. No sound is not the problem though. 
I’ll close this issue for the time being and ping back with a solution if I find it. 

Thanks for the help everyone!

closed 2020-06-13 by ArcherN9

awesomecoder3000, 2020-12-10:
I may be able to help with the connected TV having a black screen.
You may want to try added the following line to your `/boot/config.txt` file: `ignore_lcd=1`. This command tells the Raspberry Pi to ignore any LCD connected to it, and only connect to external screens.

To do this, open the terminal on your Raspberry Pi, or an SSH session from a separate device. Run the command `sudo nano /boot/config.txt`, and then, add the text `ignore_lcd=1` to any unoccupied line. In order to save your changes, press `CTRL + X`, then press` Y`. The final step is to reboot your Raspberry Pi, by unplugging and plugging it back in again, or running the command `sudo reboot`.