FD-/RPiPlay #130: build problem with fdk-aac

pyy, 2020-06-15:
Thanks @pallas and all contributors for such a good project.

For your information:

For master branch, 
After the command "cmake ..", the terminal shows **"You have called ADD_LIBRARY for library fdk-aac without any source files. This typically indicates a problem with your CMakeLists.txt file".** And the make command cannot continue.

So I check the history of `CMakeLists.txt file`
https://github.com/FD-/RPiPlay/commits/master/lib/CMakeLists.txt

This error shows on Commit on May 24, 2020 https://github.com/FD-/RPiPlay/commit/3545d5a56f82caf5dfcbb13e29aca8832030aca1#diff-5b579ba0237255aa2c66982834e4fa5e

But the earlier commit on Commits on Jul 10, 2019 https://github.com/FD-/RPiPlay/commit/3fba7078900fde6aba8a68e6779dff3da3c35473#diff-5b579ba0237255aa2c66982834e4fa5e
doesn't have this problem. So I use this commit so far.

Reference:
#124

pallas, 2020-06-15:
After that commit, you need to install `libplist-dev`.

pallas, 2020-06-15:
The message about `fdk-aac` is spurious, btw, and the `cmake` team is aware that they should no longer be warning since you can `target_sources` a library later, which is what we're doing in this code.

pyy, 2020-06-16:
After install `libplist-dev`, it works now. Thank you @pallas  :)

DaCapitalist, 2020-06-16:
Good to hear...
@pyy I suggest to close this issue then

closed 2020-06-17 by FD-