FD-/RPiPlay #131: Run script on client connect

dkasipovic, 2020-06-16:
Is it possible to execute a script on client connect? I want to automatically switch to my TV HDMI when I connect to RPiPlay with my iPhone. I have a node script that does just that. What would be the best approach?

Otherwise, great job! It runs great!

DaCapitalist, 2020-06-16:
I think @pallas posted a script that can help you already in 2 similar contexts. Does this make sense in your use case?

https://github.com/FD-/RPiPlay/issues/82#issuecomment-622058318

dkasipovic, 2020-06-17:
Yeah, this might work, that's an idea I had myself. Use lsof to see if someone is connected then run my script if yes (my script can also check if tv is already on HDMI so it would not run multiple times if not needed). This seems a bit hacky though, I just thought it might be cleaner to have onconnect and ondisconnect scripts, is all.

Thank you

pallas, 2020-06-22:
The tricky part about doing it in-process is that there are still some unresolved/hard to debug conditions that hang the daemon. The script solves those as well.

pallas, 2020-10-06:
Closing due to staleness. If the above script does not work for you, I added the enhancement tag to Issue #131.

closed 2020-12-19 by pallas