FD-/RPiPlay #135: Configure Audio

kamikaze463, 2020-06-29:
I could be sounding really stupid but I have one question, How do I route audio to 3.5mm jack instead of HDMI. Currently the audio output is just fine with HDMI.

flatsiedatsie, 2020-07-01:
I have the same question.

flatsiedatsie, 2020-07-01:
are you running it with `-a analog`?

kamikaze463, 2020-07-07:
Yes. I’m

flatsiedatsie, 2020-07-07:
Have you tried `amixer cset numid=3 1` ?

kamikaze463, 2020-07-09:
it throws an error 
"amixer: Cannot find the given element from control default"

`amixer` gives me below output:

_Simple mixer control 'HDMI',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback 400 [100%] [4.00dB] [on]_

However, I've set the playback to headphones in raspiconfig

FD-, 2020-07-09:
Since RPiPlay uses OpenMAX for audio, not ALSA, amixer won't help here, I'm afraid. The -a analog switch should work though (it does for me on Pi Zero and 3B+) !

kamikaze463, 2020-07-10:
Well, something happened but the issue seem to be fixed since yesterday's apt upgrade. Not sure if something in that update fixed my issue.

pallas, 2020-10-06:
Seems resolved, closing. Please lmk if that's incorrect.

closed 2020-10-06 by pallas