FD-/RPiPlay #137: aacDecoder_DecodeFrame error

flatsiedatsie, 2020-07-01:
From reading the issue queue I know this isn't an error, but a lack of implementation. But I thought I'd mention it anyway, as many users may be searching based on this error.

This happens when I try to use only audio casting (and not display mirroring), for example when I just want to stream some music to the Raspberry Pi 4. From what I've read, supporting audio only casting is currently not what RpiPlay should be used for. Use Shairport-Sync for that.

The good news is that Shairport Sync and RPiPlay seem fine with running alongside eachother. Another upshot is that if you want to only stream audio, the second display isn't also set to mirror the desktop.

My only issue is that it can lead to a lot of misunderstanding with end users. When they want to stream audio, they will currently see both an `RPiPLay` and `Shairport` option in the dropdown.

Would it perhaps be easy to avoid MacOS showing RPiPlay as an audio-only output option in that dropdown? Then if both pieces of software were given the same display name, users might not even realize they're connecting to two different pieces of software.

flatsiedatsie, 2020-08-26:
I just tried to comment out the line `dnssd_register_raop(dnssd, port)` in rpiplay.cc, but that didn't do it.

Maybe there are other places I should comment it out. It's mentioned in `dsnsd.c` and `dnssd.h` too. I commented out the entire function in the first, and the reference in the latter.

It still shows up as 'RpiPlay'. Very tenacious!

FD-, 2020-08-26:
Could be because a genuine Mirroring-only device doesn't exist. All official Apple products that support Mirroring also implement the other AirPlay subprotocols (audio, video, etc).

flatsiedatsie, 2020-08-26:
Yes that's what I was starting to suspect too.

pallas, 2020-10-06:
Was the actual trace ever posted?

pallas, 2020-10-06:
Sorry, did not mean to close.

pallas, 2020-10-06:
I am bad at clicking, apparently.

flatsiedatsie, 2020-10-10:
What do you mean by 'trace'?

pallas, 2020-11-22:
I have a commit at https://github.com/pallas/RPiPlay/commit/f4ef14cc63d2d5ef89127ab63ac08d3cf120bad1 that bumps to a newer-but-not-released version of FDK-AAC.

pallas, 2020-12-19:
stale, & we're tracking upstream

closed 2020-12-19 by pallas