FD-/RPiPlay #138: Set display resolution?

flatsiedatsie, 2020-07-01:
I have a 1024x600 touch screen attached to a Raspberry Pi 4.

When I use screen mirroring, my mac doesn't seem to understand what resolution the screen it's mirroring to actually is. It outputs something like full HD to it, which probably wastes bandwidth.

In display settings I was able to bring this down to 1280x720. If I press the alt key when clicking on the settings I can also select `1024x768` or `800x600`, which come close. But they're not correct, leading to wasted space when mirroring content.

Is there a way to let the server broadcast the resolution that the mirrored display actually has?

FD-, 2020-07-01:
Play around with different values in the resolution fields of [raop_handlers' info method](https://github.com/FD-/RPiPlay/blob/6eeb5bae02dfdbac38ceecd63574d8d480687058/lib/raop_handlers.h#L28). If you find putting the correct values there works, we need to figure out how a program on the Pi can get the monitor resolution. Would getting the framebuffer dimensions work?

flatsiedatsie, 2020-07-12:
I've created a PR for some changes I made to allow users to set the prefered resolution via a parameter. Unfortunately, setting it in RAOPhandler's info method doesn't seem to always set it to that resolution. Maybe there's some other factor at work?

Big-Sur, 2020-08-23:
I'm also very interested in fiddling with native or customizable resolutions. Can anyone point this noob in the direction of where to begin?

flatsiedatsie, 2020-08-26:
For now you could use my branch as a starting point: https://github.com/flatsiedatsie/RPiPlay

pallas, 2020-10-06:
Could you please turn this into a PR?

DrKittens, 2020-10-07:
Additionally capability to add stretching of the mirrored application/overscan percentages in would be desireable.

From the first post i think this issue is more to do with aspect ratios as 1024x600 is a 4:3 ratio and 1280x720 is a 16:9 ratio which apple devices (ipad/iphone etc) all output (okay some are 16:10 but thats alot less picturebox than 4:3 and 21:9/32:9)

When using an ultrawide/4:3 monitor as the display a significant amount of the screen is black bars, bumping the resolution in rpiplay doesnt do anything as the ipad/macbook/whatever simply has a different aspect ratio and wont/cant transform the image into a different aspect ratio.

**TLDR:**I believe the actual issue in the OP is to do with underscanning not the resolution configured in rpiplay

DrKittens, 2020-10-07:
Options to configure an application overscan/forced stretching/arbitrary scaling to handle monitors of different aspect ratios would also be a welcome enhancement.

Using flatsiedatsie's fork and configuring a 21:9 resolution to display has no effect.

pallas, 2020-10-09:
Sorry, I didn't see it referenced!

Zenexer, 2020-12-15:
If I modify these four lines to get 1280x1024:

```
    plist_t displays_0_width_node = plist_new_uint(1280);
    plist_t displays_0_height_node = plist_new_uint(1024);
    plist_t displays_0_width_pixels_node = plist_new_uint(1280);
    plist_t displays_0_height_pixels_node = plist_new_uint(1024);
```

macOS reports 1280x1024 @ 30 Hz:

[image]

and my monitor reports it's receiving a signal at 1280x1024 @ 60Hz.

However, the actual image displayed on the monitor is quite blurry, with most text being illegible.

It seems a bit odd that macOS is showing two devices for the same physical display, and that the refresh rate doesn't match, but those are probably unrelated to the issue.

Is this likely to be related to my resolution modifications, or is it more likely to be an unrelated issue?

flatsiedatsie, 2020-12-15:
@Zenexer Out of curiosity, would you be willing to test if you can get MacOS to recognise the display as `1024x600`? If so, then that would be good news.

Zenexer, 2020-12-15:
@flatsiedatsie I’ll try, but I won’t actually be able to tell whether it’s working or has the same bug, since it doesn’t match the resolution of my monitor. macOS recognizes it just fine currently, but it looks like it’s outputting at a lower resolution behind-the-scenes.

Zlitus, 2021-02-28:
Hello, thank you very much for your amazing work.
I need something similar but for my 4K display, I tried to change it like that from the current master:
```
    plist_t displays_0_width_node = plist_new_uint(3840);
    plist_t displays_0_height_node = plist_new_uint(2160);
    plist_t displays_0_width_pixels_node = plist_new_uint(3840);
    plist_t displays_0_height_pixels_node = plist_new_uint(2160);
```

And macOS regognize it fine, and the connection is established, but the screen stay black (No errors on the console, except than we stay stuck on “Stopping” when CTRL+C).

I tried with gstreamer & rpi renderer, same result. Even if in macOS I put back 1920x1080 for the second screen, it stay black. Even if my 1st connection is on 1920x1080, it stay black 😅. I'm using macOS 11.2.2 & Raspbian Lite 10.

Thank you for your help!

[image]

FD-, 2021-02-28:
@Zlitus 4K will never work, for the simple reason that the Pi can only HW decode H264 up to 1080p@30, 50fps if overclocked. The H264 decode block in the SoC has not been changed since the first Pis. Additionally, I'm not sure if 4K is supported by the AirPlay protocol at all. Someone would have to test this on an AppleTV 4K.

Zlitus, 2021-02-28:
Thank you very much for your answer 👍. Just after my message I had an idea, I thought than maybe, the problem was actually not related to RPiPlay but to a bad detection of my screen by raspbian or something. So I tried the same thing with Ubuntu Desktop and it is actually working, but you're right: It is very slow (I would say, watching the clock on macOS, than every second I got 2sec of delay in the airplay-mirrored 4K).
I think the AppleTV 4K can do a 4K Airplay, but I don't have one to check.

Zlitus, 2021-03-01:
And just another related question, I saw than the Pi4b is capable of decoding 4K video at 60fps when we use a h265 stream, and for what I see, Apple on recent AppleTV now use h265 streams. Do you think it would be possible to make RPiPlay compatible with h265?

jankais3r, 2021-03-13:
I have a display with native resolution of 2560x1600. When I am mirroring the user interface of iOS from my iPad, all is fine (that is, full height of the display is used, with black bars on both sides). However, when I play a movie from Photos app, rpiplay switches to 1920x1080 mode which in turn takes full width of the display but puts black bars on top and bottom. As the screen's native width is 2560, this also means that the image is upscaled from 1920. I would like rpiplay to utilize the screen's full native resolution when in this "playing a video" mode of mirroring.

I can see the switching of the mode also in the log:
```
$ rpiplay -n Test -l -d
Using low-latency mode
> stream info: channel = 2	sample_rate = 44100	frame_size = 480	aot = 39	bitrate = 0
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.1.69
Remote: 192.168.1.23
httpd receiving on socket 15
conn_request
Handling request GET with URL /info
INFO len = -1286629116
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request SETUP with URL rtsp://192.168.1.69/1251772004594763452
DACP-ID: F11E9C8BE06DA323
Active-Remote: 2230650721
Transport: null
SETUP 1
eiv_len = 16
ekey_len = 72
fairplay_decrypt ret = 0
timing_rport = 51663
raop_ntp parse remote ip = 192.168.1.23
raop_ntp starting time
raop_rtp parse remote ip = 192.168.1.23
raop_rtp_mirror parse remote ip = 192.168.1.23
eport = 40675, tport = 45789
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = -1614783717361590
Accepted IPv4 client on socket 17
Local: 192.168.1.69
Remote: 192.168.1.23
httpd receiving on socket 15
conn_request
Handling request GET with URL /info
INFO len = -1286629116
httpd receiving on socket 15
conn_request
Handling request GET_PARAMETER with URL rtsp://192.168.1.69/1251772004594763452
httpd receiving on socket 15
conn_request
Handling request RECORD with URL rtsp://192.168.1.69/1251772004594763452
raop_handler_record
httpd receiving on socket 15
conn_request
Handling request SET_PARAMETER with URL rtsp://192.168.1.69/1251772004594763452
httpd receiving on socket 15
[... 122 lines cut ...]
conn_request
Handling request SET_PARAMETER with URL rtsp://192.168.1.69/5286778094632452153
httpd receiving on socket 15
conn_request
Handling request SETUP with URL rtsp://192.168.1.69/5286778094632452153
DACP-ID: 402F2D08891EAA37
Active-Remote: 4283404641
Transport: null
type = 110
streamConnectionID = 9296301987114010920
raop_rtp_mirror starting mirroring
Mirroring initialized successfully
raop_rtp_mirror accepting client
raop_rtp_mirror width_source = 2062.000000 height_source = 1440.000000 width = 2154.000000 height = 1504.000000
raop_rtp_mirror sps size = 18
raop_rtp_mirror pps size = 4
Got h264 data of 30 bytes
Injecting max_dec_frame_buffering
Video delay is 1615667444829164
raop_rtp_mirror video ntp = 1615667444849471, now = 1615667444831477, latency = -17994
Got h264 data of 22668 bytes
Video delay is -17131
```

Any idea what could be causing this, and how to get full native resolution of the display when playing videos from the Photos app?

jankais3r, 2021-03-19:
After some more experimentation, I discovered that portrait mirroring devices in portrait orientation works even when RPiPlay is compiled for the 2560x1600 display.

Works:
- Portrait iPhone
- Portrait iPad

Doesn't work:
- Landscape iPhone
- Landscape iPad
- Photos/videos played from Photos app

Could there be some sort of video buffer that is overflowing? The mirrored video in landscape mode does not have to be scaled down as much compared to the portrait orientation, meaning that more non-black pixels have to be rendered, potentially causing this issue.

Video here: https://www.youtube.com/watch?v=YQ20BlQ7ulg

mjsj12, 2023-11-09:
>     plist_t displays_0_width_node = plist_new_uint(3840);
>     plist_t displays_0_height_node = plist_new_uint(2160);
>     plist_t displays_0_width_pixels_node = plist_new_uint(3840);
>     plist_t displays_0_height_pixels_node = plist_new_uint(2160);

hi,

On which file are these configurations modified?