FD-/RPiPlay #139: Starting

Jack-Barnard2020, 2020-07-05:
Hello,
Sorry i may just being stupid but i run the final command 'make' then it does its thing no errors come up but when i go on my iPhone and got to screen mirroring in the control centre nothing shows up. is there a start command not listed?

urbanvanilla, 2020-07-05:
Usage
Start the rpiplay executable and an AirPlay mirror target device will appear in the network. At the moment, these options are implemented:

-n name: Specify the network name of the AirPlay server.

-b (on|auto|off): Show black background always, only during active connection, or never.

-r (90|180|270): Specify image rotation in multiples of 90 degrees.

-l: Enables low-latency mode. Low-latency mode reduces latency by effectively rendering audio and video frames as soon as they are received, ignoring the associated timestamps. As a side effect, playback will be choppy and audio-video sync will be noticably off.

-a (hdmi|analog|off): Set audio output device

-d: Enables debug logging. Will lead to choppy playback due to heavy console output.

-v/-h: Displays short help and version information.

Jack-Barnard2020, 2020-07-07:
Sorry that doesn’t help how do you start the program

urbanvanilla, 2020-07-08:
This is pretty basic stuff but what you need to do is run the rpiplay executable from a terminal window.
e.g. type ./rpiplay wherever you have the executable. or if running ti from by default it'll be "./RPiPlay/build/rpiplay" and you can add on the commands in the instructions after to use the options
e.g "./RPiPlay/build/rpiplay -n JacksRpiplayserverthing "

Jack-Barnard2020, 2020-07-08:
Thanks it’s just I am new to it all and it just a bit confusing cause I’ve only really done python and c++

closed 2020-07-09 by Jack-Barnard2020