FD-/RPiPlay #144: can you add a reflect (or flip) function?

jak543, 2020-07-25:
Hello! I have a raspberry pi 4b,because the offical updated their configuation(https://www.raspberrypi.org/documentation/configuration/config-txt/video.md),display_rotate can't use on 4b,I use the xrandr command to rotate and reflect the screen,but when I start your RPiPlay and have one device linked,picture become to the original direction,so i need to flip both X and Y axis to get the right picture,can you add this function?

FD-, 2020-07-25:
I assume you mean to rotate the image, not flip it? If so, just use the -r command line argument.

jak543, 2020-07-25:
no,I flipped my system screen using xrandr command(xrandr --output [mymonitername] --reflect x) becase I made something like google glass,the desktop looks good, but your RPiPlay were still in the original direction,it didn't change as the desktop flipped

so can you add horizontal flip and vertical flip in RPiPlay

FD-, 2020-07-25:
Yeah, RPiPlay renders on the GPU directly using OpenMAX, so xrandr doesn't work here. However, it seems the video_render OpenMAX component has an OMX_IndexConfigCommonMirror property that can be used for implementing what you want. Commit f387248b0427122979d9786e0367312fe30ce6f5 (the rotation implementation) should be a good starting point. If you have something that could work for others as well, please submit a PR!

garethmarlow, 2020-08-30:
This would be great - I'd be really grateful if this could be implemented. I've built a teleprompter to use with Zoom calls from my mac using an old 19" Samsung 4:3 LCD display. It's currently directly connected to my macbook pro but the image is reflected in the display, so I need to L-R flip it.

I haven't been able to find a way to do this from the mac but Airplay using RPiPlay would solve the problem if the display can be reversed.

dougg3, 2020-09-07:
I just followed the template of the rotation commit above and added horizontal/vertical flipping support. See pull request #154 if you want to test it out! Seems to work fine from what I can see.

pallas, 2020-10-06:
This was merged, closing.

closed 2020-10-06 by pallas