FD-/RPiPlay #145: CMake Error

evaspaceman, 2020-07-26:
I received the following error when executing cmake.

Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.1.1d")
CMake Error: The following variables are used in the project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBPLIST
     linked by target "airplay" in directory /home/pi/RPiPlay/lib

-- Configuring incomplete, errors occurred!

Any help getting beyond this would be appreciated.

jak543, 2020-07-26:
I moved the project to the /usr/local/lib and it can be cmake ..,I tried to add the path in the cmakelist.txt but it didn't work,so I moved it

evaspaceman, 2020-07-26:
Is there something I need to do to execute “cmake ..” without error?  With this error, the cmake fails and stops.   I am running an updated/clean version of Raspian Buster.

FD-, 2020-07-26:
I know build errors often appear discouragingly cryptic to beginners, but this particular one should be pretty easy to understand with just a little bit of common sense: It’s mentioning libplist, which implies the problem is related to a) a library (“lib”), and b) a particular library that is mentioned in the prerequisites of RPiPlay. So, the natural thing to do would be make sure you installed all dependency libraries. You’ll very likely discover you didn’t install libplist, which you can do by using the command `sudo apt-get install libplist-dev`, as mentioned in the instructions.

evaspaceman, 2020-07-26:
That was it. Thank you.  I appreciate your help.

closed 2020-07-27 by FD-