FD-/RPiPlay #147: RPiPlay Freezing

FlashVaultAdmin, 2020-07-31:
RPiPlay keeps freezing when using as a separate display on macOS Big Sur. Tried to drag a tab with a YouTube video playing on it, and RPiPlay froze immediately. Tried rebooting, but that didn't work. After the reboot, I tried dragging the same tab to the RPiPlay screen, and it froze again.

Using a RPI 3B, not overclocked

FlashVaultAdmin, 2020-07-31:
Also, is there any way to customize the black screen when RPiPlay is waiting for a connection?

FlashVaultAdmin, 2020-07-31:
Sorry for all the comments, but there is another problem. When I try to play a video with audio, the audio cuts off every 500 milliseconds, so much so that the audio is barely recognizable.

FlashVaultAdmin, 2020-08-01:
Is this project discontinued?

pallas, 2020-08-01:
No, it is not discontinued. I'm on my phone right now, but a few months ago I posted a script that allows you to put other things on the screen until someone connects. As for audio, I assume you're not running in low latency mode. There are known issues with audio/video sync at the moment. Please try low latency to see if it helps.

FlashVaultAdmin, 2020-08-02:
Thanks. When I put it in low-latency mode, the colors shown look like nothing that is on the iPad, or when RPiPlay is not in low-latency mode. What about RPiPlay freezing? The freezing now happens after a few seconds after any device connects, and when I disconnect my device, RPiPlay still shows the screen where it froze.

pallas, 2020-08-02:
Here's the script I mentioned. https://github.com/FD-/RPiPlay/issues/82#issuecomment-622058318

I'm running this directly via xinit, not in a terminal or something under X. RPiPlay sends h264 directly to the GPU.. are you sure something else isn't trying to write to the HDMI at the same time? My wild guess would be that your GPU is getting hung because something is trying to use it at the same time RPiPlay is but without more information about your setup I have no idea.

I'm running on a 3B+ (not a 3B) and it works fine, but I did also enable GL and increased available video memory. YMMV.

Did you compile in Release mode with the suggested CFLAGS/CXXFLAGS?

The branch I'm running is at https://github.com/pallas/RPiPlay/tree/running-branch which includes a couple of additional FDK-AAC fixes and disables an unsafe h264 parser hack.

FlashVaultAdmin, 2020-08-02:
About the script: How exactly do I use it? I want to put a custom image made by me on the screen. Sorry if I am asking much!

About the freezing: As I said before, I am using a RPI 3B (not CPU/GPU overclocked, and no memory change for GPU). I am running stock raspbian cli (no desktop), so the only program that is installed in RPiPlay. I installed RPiPlay using the commands provided, and I did not add any additional flags to the make command. The only thing in my mind that could be using the GPU is the raspbian cli interface, even though that wouldn't make sense. Could you provide me with the commands that has the flags. I'll load a new raspbian cli onto the SD Card, and I'll run the new commands. I'll enable GL and increase video memory to 3G/1G.

FlashVaultAdmin, 2020-08-03:
About the script: How exactly do I use it? I want to put a custom image made by me on the screen. Sorry if I am asking much!

About the freezing: Found the command for make with the O3 flags. I'll try it with GL and 3G/1G memory spilt.

FlashVaultAdmin, 2020-08-03:
About the script: How exactly do I use it? I want to put a custom image made by me on the screen. Sorry if I am asking much!

About the freezing: I compiled RPiPlay with the O3 flag and I dedicated 512MB (Half of the RPI 3B's RAM) to GPU Memory. It is still freezing after a few seconds of someone connecting. If you are wondering about GL, I tried to turn it on, but that only resulted in the mirroring part of RPiPlay not even showing, so I turned it off. Anything else I can try?

FlashVaultAdmin, 2020-08-04:
Hi. It's been 2 days since anyone responded to any of my questions. Can I know what's happening?

FD-, 2020-08-04:
This is a free open-source project, which means we're developing it voluntarily in our own free time. You're free to use and modify the code to your liking, but obviously, all maintainers are still allowed to freely decide on the amount of their time they want to invest. For me at least, this is a low priority side project, and I'm not willing to investigate a bug that I don't see on my system. That being said, you have access to the whole source code, so if you feel like having the problem fixed is worth your time, please go ahead! I greatly appreciate any contribution!

FlashVaultAdmin, 2020-08-05:
I can show you what's happening though a video, if that's what you want

pallas, 2020-08-05:
tl;dr, since I knocked out the (very few) bugs I was seeing, I mostly only have time to look at this on weekends. I have my RPi set up on my TV the way I want and I've been thinking a bit about how to fix the sync issue, but no time to investigate. I *am* actively tracking http-parser & fdk-aac but that's about it.

To answer the question about the script, called it `/usr/lib/kiosk` and run `xinit /usr/lib/kiosk &` in `/etc/rc.local`

FlashVaultAdmin, 2020-08-05:
So to fix the freezing issue, I should get a rpi 4?

pallas, 2020-08-05:
I'm running on a 3B+ and it's not freezing. Did you try my upstream branch re: fdk-aac?

FlashVaultAdmin, 2020-08-05:
You mean your branch or RPiPlay? No, I didn't try that yet. I'll try it!

FD-, 2020-08-05:
BTW, the freezing issue could also be caused by macOS Big Sur, which AFAIK is still in beta. I haven't tried it (my only mac is my daily driver), but I suggest you test with a stable macOS version.

FlashVaultAdmin, 2020-08-05:
I highly doubt its a macOS Big Sur issue, since the freezing is also happening on my iPad which is on 13.5.1, but as pallas said, I'll try RPiPlay with his repo.

FlashVaultAdmin, 2020-08-05:
I'll also post a video if it happens with his repo too.

pallas, 2020-08-06:
LMK if it helps

pallas, 2020-08-06:
FYI, there was a small window of time when my branch was broken last night because I was updating http-parser.

closed 2020-09-08 by FlashVaultAdmin