FD-/RPiPlay #148: CMake Error: The following variables are used in this project, but they are set to NOTFOUND.

ParaDescartar123, 2020-08-07:
Getting this below.

Followed the instructrions and searched for similar issues.  There is one, but there is no solution. Just says to do it outside a docker build, but I am not on Docker. I am on Ubuntu.

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
BCM_HOST
    linked by target "ilclient" in directory /home/ubuntu/RPiPlay/renderers
BRCM_EGL
    linked by target "ilclient" in directory /home/ubuntu/RPiPlay/renderers
BRCM_GLES_V2
    linked by target "ilclient" in directory /home/ubuntu/RPiPlay/renderers
OPENMAXIL
    linked by target "ilclient" in directory /home/ubuntu/RPiPlay/renderers
VCHIQ_ARM
    linked by target "ilclient" in directory /home/ubuntu/RPiPlay/renderers
VCOS
    linked by target "ilclient" in directory /home/ubuntu/RPiPlay/renderers

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/RPiPlay/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/RPiPlay/build/CMakeFiles/CMakeError.log".

pallas, 2020-08-07:
Those are the system libraries for Raspberry Pi OS. Currently the project only works on RPis, although there is a pull request to enable other systems via gstreamer.

~Derrick • iPhone

pallas, 2020-10-06:
Stale for a month, closing.

closed 2020-10-06 by pallas