FD-/RPiPlay #150: Docker builder

nicolaspernoud, 2020-08-17:
Hello,

Thanks for this amazing work.
For your information, I made a script that build RPiPlay with docker and enable it to be used directly in LibreELEC (by getting the libdns_sd.so.1 library locally). Feel free to include it in your repo or point to it in your readme...

Best regards.

pallas, 2020-10-06:
Thanks, this is great. If you want to include it in mainline, can you please make a PR?

closed 2020-10-06 by pallas

jhfdevelop, 2020-10-22:
Hey,
thank you everyone for your great work.
I attempted to containerize the app for the raspberry pi using your docker builder @nicolaspernoud.

Feel free to have a look: https://github.com/jhfdevelop/rpiplay_docker

nicolaspernoud, 2020-10-24:
@jhfdevelop : I had a look, nice !
@pallas : PR still pending...