FD-/RPiPlay #156: Blue Screen 10 min after dair play mirroring starts

vulwerine, 2020-09-09:
Hey .... I’d first like to thank you for this. 
I’m using a RPi-4 running raspian 
I’ve connected the RPi to my TV and it is running smoothly but exactly 10min after the mirroring starts I get a blue screen(video signal is cut off). 
Probably my pi is turned off, I don’t know. I can’t end the program or go back to the terminal. I’ve to switch off power and start it back again.
Please help me out. 
Thanks ....

pallas, 2020-09-09:
Are you sure it's not overheating?

gorbypark, 2020-09-18:
@vulwerine It's possible that your power supply isn't providing enough current.  I've had a number of issues that I couldn't figure out magically disappear after getting a better power supply.

vulwerine, 2020-09-19:
> @vulwerine It's possible that your power supply isn't providing enough current. I've had a number of issues that I couldn't figure out magically disappear after getting a better power supply.

Thanks a lot. I switched to a 18W power brick and it worked. I don’t have that problem anymore.

closed 2020-09-19 by vulwerine