FD-/RPiPlay #162: Connected and working, but windowed, not fullscreen

ema07cqc, 2020-09-29:
Hello there, this is a great project.
Got it working on Ubuntu 20.04 but its windowed mode. There is a way to get it fullscreen? Thanks,

rajb245, 2020-10-02:
@ema07cqc can you share more about how you got it working on Ubuntu? I used apt to install all the pre-requisites, was able to cmake and make, and call it like this: `./rpiplay -n "Laptop" -d`
I can see an AirPlay device called "Laptop" on my iPhone and it connects. I can even hear the audio come across the connection! But I don't see any video. Does rpiplay open a new window for the video automatically when it starts, or when something connects?

ema07cqc, 2020-10-02:
> @ema07cqc can you share more about how you got it working on Ubuntu? I used apt to install all the pre-requisites, was able to cmake and make, and call it like this: `./rpiplay -n "Laptop" -d`
> I can [...]

Hi there. I tried mirroring my MacBook 12" with Catalina 10.15.7.
Your command seems ok. Im using "/RPiPlay/build/./rpiplay -l -a off -b off" because i don't need audio.
It opens a new window titled "unknown" and I can o minimize or resize only.

RafaelGreenblatt, 2020-10-03:
I was able to do this by setting a shortcut for "toggle fullscreen mode" in the "keyboard" section of system settings.  AFAICT this is an all-purpose solution (i.e. nothing to do specifically with this project).

pallas, 2020-10-06:
Those instructions are helpful. Since it's general and it sounds like it works, going to close this issue.

closed 2020-10-06 by pallas

tanmaster, 2022-05-21:
> I was able to do this by setting a shortcut for "toggle fullscreen mode" in the "keyboard" section of system settings. AFAICT this is an all-purpose solution (i.e. nothing to do specifically with this project).

this worked for me as well