FD-/RPiPlay #164: S/PDIF

emanuelelaface, 2020-10-01:
Hello,
I'd like to use RPiPlay on a Linux box that has a S/PDIF optical audio. I compiled RPiPlay but the two options HDMI and Analog do not produce any sound on the S/PDIF (I tried and I can have sound correctly on the HDMI output and on the analog output, but not on the optical S/PDIF).

Do you think that is possible to add this feature to the code?
Many thanks!

emanuelelaface, 2020-10-02:
I have an update on this front. If I try to play a slideshow with photos I have no video, but the background music of the slideshow plays correctly on the RPiPlay server through the S/PDIF optical.
So I think that after all there is everything in the code to work, it is just a matter to specify that the audio has to use the same route.

pallas, 2020-10-06:
Please let us know if you are able to figure it out. Since you think it's a configuration issue unrelated to this project, I'm going to close it out. However, knowing what the configuration should be could help other users with the same setup.

closed 2020-10-06 by pallas