FD-/RPiPlay #166: Video dosen’t work on raspberry pi os 64 bit

orisystech, 2020-10-02:
I installed rpiplay on raspberry pi os 64 bit. I tried to do the mirroring and I noticed that only audio works, the video is not transmitted even if in the terminal it is indicated that the connection is established..

pallas, 2020-10-06:
Are you sure you've pinned it to the right HDMI? Which model RPi do you have?

orisystech, 2020-10-06:
I have a Pi 3 B+ and I run rpiplay with option “-b auto -a hdmi -l”.

orisystech, 2020-10-08:
I tried to run RpiPlay also on DietPi 64 bit but doesn’t work. I think I’ve tried in the past even with a version of archpi 64 bit and I seem to remember it worked. Maybe the issue is 64 bit library of distro debian based.

dougg3, 2020-10-12:
I don't actually have a physical Raspberry Pi that can run the 64-bit OS, but I played around a little bit using qemu on the [latest beta of Raspberry Pi OS (64-bit)](https://www.raspberrypi.org/forums/viewtopic.php?t=275370) and it looks like several of the dependencies aren't yet part of the beta 64-bit OS. Maybe that could be playing a part in it?

When you run the cmake command, are you seeing this message? I am seeing it:
`OpenMAX libraries not found, skipping compilation of Raspberry Pi renderer`

Developer info: I printed out the results of the CMake find_library calls and got this:
```
BRCM_GLES_V2-NOTFOUND
BRCM_EGL-NOTFOUND
OPENMAXIL-NOTFOUND
/usr/lib/aarch64-linux-gnu/libbcm_host.so
/usr/lib/aarch64-linux-gnu/libvcos.so
/usr/lib/aarch64-linux-gnu/libvchiq_arm.so
```

There are [other projects that are reporting issues](https://github.com/jellyfin/jellyfin/issues/4023) due to the lack of libopenmaxil on 64-bit.

If you run:

`./rpiplay -h`

...which renderers are listed under the audio and video renderers? With the default Raspberry PI OS 64-bit, I only see this, which would result in nonworking video *and* audio:

```
-vr renderer          Set video renderer to use. Available renderers:
    dummy: Dummy renderer; does not actually display video [Default]
-ar renderer          Set audio renderer to use. Available renderers:
    dummy: Dummy renderer; does not actually play audio [Default]
```

I would assume the reason RPiPlay isn't working is because libbrcmGLESv2.so, libbrcmEGL.so, and libopenmaxil.so aren't currently built for the 64-bit Raspberry Pi OS. Maybe audio works because it's using the gstreamer audio renderer instead?

SumOlgy, 2020-12-08:
I’ve just compiled this and while I can see and connect to the Airplay device I’m not seeing any video output on either of the RPi4’s HDMI outputs.  Any help/tips appreciated

                                                                                                                                                                   ubuntu@ubuntu:~$ rpiplay -h                                                                                                                                                                                                                                                                                                                                                                                                                                                                              ubuntu@ubuntu:~$ rpiplay -h                                                                                                                                        RPiPlay 1.2: An open-source AirPlay mirroring server for Raspberry Pi                                                                                              Usage: rpiplay [-n name] [-b (on|auto|off)] [-r (90|180|270)] [-l] [-a (hdmi|analog|off)] [-vr renderer] [-ar renderer]                                            Options:                                                                                                                                                           -n name               Specify the network name of the AirPlay server                                                                                               -b (on|auto|off)      Show black background always, only during active connection, or never                                                                        -r (90|180|270)       Specify image rotation in multiples of 90 degrees                                                                                            -f (horiz|vert|both)  Specify image flipping (horiz = horizontal, vert = vertical, both = both)                                                                    -l                    Enable low-latency mode (disables render clock)                                                                                              -a (hdmi|analog|off)  Set audio output device                                                                                                                      -vr renderer          Set video renderer to use. Available renderers:                                                                                                  gstreamer: GStreamer H.264 renderer [Default]                                                                                                                      dummy: Dummy renderer; does not actually display video                                                                                                         -ar renderer          Set audio renderer to use. Available renderers:                                                                                                  gstreamer: GStreamer audio renderer [Default]                                                                                                                      dummy: Dummy renderer; does not actually play audio                                                                                                            -d                    Enable debug logging                                                                                                                         -v/-h                 Displays this help and version information                                                                                                   ubuntu@ubuntu:~$ uname -a                                                                                                                                          Linux ubuntu 5.4.0-1023-raspi #26-Ubuntu SMP PREEMPT Thu Nov 12 14:58:33 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux                                                ubuntu@ubuntu:~$

pallas, 2020-12-08:
If you install gstreamer's dev libraries and re-cmake + recompile, do you see those renderers?

SumOlgy, 2020-12-08:
Sorry - I pasted some Information and it didn't get saved.

It's running Ubuntu 20.04 LTS - -all updated.  
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 5.4.0-1023-raspi #26-Ubuntu SMP PREEMPT Thu Nov 12 14:58:33 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux

ubuntu@ubuntu:~$ rpiplay -h
RPiPlay 1.2: An open-source AirPlay mirroring server for Raspberry Pi
Usage: rpiplay [-n name] [-b (on|auto|off)] [-r (90|180|270)] [-l] [-a (hdmi|analog|off)] [-vr renderer] [-ar renderer]
Options:
-n name               Specify the network name of the AirPlay server
-b (on|auto|off)      Show black background always, only during active connection, or never
-r (90|180|270)       Specify image rotation in multiples of 90 degrees
-f (horiz|vert|both)  Specify image flipping (horiz = horizontal, vert = vertical, both = both)
-l                    Enable low-latency mode (disables render clock)
-a (hdmi|analog|off)  Set audio output device
-vr renderer          Set video renderer to use. Available renderers:
    gstreamer: GStreamer H.264 renderer [Default]
    dummy: Dummy renderer; does not actually display video
-ar renderer          Set audio renderer to use. Available renderers:
    gstreamer: GStreamer audio renderer [Default]
    dummy: Dummy renderer; does not actually play audio
-d                    Enable debug logging
-v/-h                 Displays this help and version information

I've tried both HDMI outputs - the cable is ok as I can see boot info from the Pi.  

ubuntu@ubuntu:~$ rpiplay -n RPIPlay
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.101.114
Remote: 192.168.101.55
Accepted IPv4 client on socket 22
Local: 192.168.101.114
Remote: 192.168.101.55
raop_rtp_mirror starting mirroring
^CStopping...
Removing connection for socket 15
Destroying connection
Removing connection for socket 22
Destroying connection

SumOlgy, 2020-12-08:
ubuntu@ubuntu:~$ sudo apt list | grep gstreamer

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

baresip-gstreamer/focal 0.6.1-1build4 arm64
gir1.2-gstreamer-1.0/focal,now 1.16.2-2 arm64 [installed,automatic]
gnome-twitch-player-backend-gstreamer-cairo/focal 0.4.1-3 arm64
gnome-twitch-player-backend-gstreamer-clutter/focal 0.4.1-3 arm64
gnome-twitch-player-backend-gstreamer-opengl/focal 0.4.1-3 arm64
gstreamer-qapt/focal 3.0.5-1ubuntu1 arm64
gstreamer1.0-adapter-pulseeffects/focal 4.7.1-2 arm64
gstreamer1.0-alsa/focal 1.16.2-4 arm64
gstreamer1.0-autogain-pulseeffects/focal 4.7.1-2 arm64
gstreamer1.0-clutter-3.0/focal 3.0.27-1 arm64
gstreamer1.0-convolver-pulseeffects/focal 4.7.1-2 arm64
gstreamer1.0-crystalizer-pulseeffects/focal 4.7.1-2 arm64
gstreamer1.0-doc/focal 1.16.2-2 all
gstreamer1.0-espeak/focal 0.5.0-1 arm64
gstreamer1.0-gl/focal,now 1.16.2-4 arm64 [installed,automatic]
gstreamer1.0-gtk3/focal 1.16.2-1ubuntu2 arm64
gstreamer1.0-libav-dbg/focal 1.16.2-2 arm64
gstreamer1.0-libav/focal,now 1.16.2-2 arm64 [installed]
gstreamer1.0-nice/focal 0.1.16-1 arm64
gstreamer1.0-omx-bellagio-config/focal 1.16.2-1 arm64
gstreamer1.0-omx-generic-config/focal 1.16.2-1 arm64
gstreamer1.0-omx-generic/focal 1.16.2-1 arm64
gstreamer1.0-opencv/focal 1.16.2-2.1ubuntu1 arm64
gstreamer1.0-packagekit/focal-updates,focal-security 1.1.13-2ubuntu1.1 arm64
gstreamer1.0-pipewire/focal 0.2.7-1 arm64
gstreamer1.0-plugins-bad-dbg/focal 1.16.2-2.1ubuntu1 arm64
gstreamer1.0-plugins-bad-doc/focal 1.16.2-2.1ubuntu1 all
gstreamer1.0-plugins-bad/focal,now 1.16.2-2.1ubuntu1 arm64 [installed]
gstreamer1.0-plugins-base-apps/focal 1.16.2-4 arm64
gstreamer1.0-plugins-base-dbg/focal 1.16.2-4 arm64
gstreamer1.0-plugins-base-doc/focal 1.16.2-4 all
gstreamer1.0-plugins-base/focal,now 1.16.2-4 arm64 [installed,automatic]
gstreamer1.0-plugins-good-dbg/focal 1.16.2-1ubuntu2 arm64
gstreamer1.0-plugins-good-doc/focal 1.16.2-1ubuntu2 all
gstreamer1.0-plugins-good/focal 1.16.2-1ubuntu2 arm64
gstreamer1.0-plugins-rtp/focal 1.14.4.1 arm64
gstreamer1.0-plugins-ugly-dbg/focal 1.16.2-2build1 arm64
gstreamer1.0-plugins-ugly-doc/focal 1.16.2-2build1 all
gstreamer1.0-plugins-ugly/focal 1.16.2-2build1 arm64
gstreamer1.0-pocketsphinx/focal 0.8.0+real5prealpha+1-6ubuntu4 arm64
gstreamer1.0-pulseaudio/focal 1.16.2-1ubuntu2 arm64
gstreamer1.0-python3-dbg-plugin-loader/focal 1.16.2-2 arm64
gstreamer1.0-python3-plugin-loader/focal 1.16.2-2 arm64
gstreamer1.0-qt5/focal 1.16.2-1ubuntu2 arm64
gstreamer1.0-rtsp-dbg/focal 1.16.2-3 arm64
gstreamer1.0-rtsp/focal 1.16.2-3 arm64
gstreamer1.0-tools/focal 1.16.2-2 arm64
gstreamer1.0-vaapi-doc/focal 1.16.2-2 all
gstreamer1.0-vaapi/focal,now 1.16.2-2 arm64 [installed]
gstreamer1.0-x/focal 1.16.2-4 arm64
libcanberra-gstreamer/focal 0.30-7ubuntu1 arm64
libgstreamer-gl1.0-0/focal,now 1.16.2-4 arm64 [installed,automatic]
libgstreamer-ocaml-dev/focal 0.3.0-2build1 arm64
libgstreamer-ocaml/focal 0.3.0-2build1 arm64
libgstreamer-opencv1.0-0/focal 1.16.2-2.1ubuntu1 arm64
libgstreamer-plugins-bad1.0-0/focal,now 1.16.2-2.1ubuntu1 arm64 [installed,automatic]
libgstreamer-plugins-bad1.0-dev/focal 1.16.2-2.1ubuntu1 arm64
libgstreamer-plugins-base1.0-0/focal,now 1.16.2-4 arm64 [installed,automatic]
libgstreamer-plugins-base1.0-dev/focal,now 1.16.2-4 arm64 [installed]
libgstreamer-plugins-good1.0-0/focal,now 1.16.2-1ubuntu2 arm64 [installed,automatic]
libgstreamer-plugins-good1.0-dev/focal 1.16.2-1ubuntu2 arm64
libgstreamer1-perl/focal 0.003-3 all
libgstreamer1.0-0-dbg/focal 1.16.2-2 arm64
libgstreamer1.0-0/focal,now 1.16.2-2 arm64 [installed,automatic]
libgstreamer1.0-dev/focal,now 1.16.2-2 arm64 [installed]
libgstreamerd-3-0/focal 3.9.0-3build2 arm64
libgstreamerd-3-dev/focal 3.9.0-3build2 arm64
libgstreamermm-1.0-1/focal 1.10.0+dfsg-2build1 arm64
libgstreamermm-1.0-dev/focal 1.10.0+dfsg-2build1 arm64
libgstreamermm-1.0-doc/focal 1.10.0+dfsg-2build1 all
libqt5gstreamer-1.0-0/focal 1.2.0-5 arm64
libqt5gstreamer-dev/focal 1.2.0-5 arm64
libqt5gstreamerquick-1.0-0/focal 1.2.0-5 arm64
libqt5gstreamerui-1.0-0/focal 1.2.0-5 arm64
libqt5gstreamerutils-1.0-0/focal 1.2.0-5 arm64
libreoffice-avmedia-backend-gstreamer/focal-updates 1:6.4.6-0ubuntu0.20.04.1 arm64
librust-gstreamer-audio-sys-dev/focal 0.8.0-1 arm64
librust-gstreamer-base-sys-dev/focal 0.8.0-1 arm64
librust-gstreamer-sys-dev/focal 0.8.0-1 arm64
librust-gstreamer-video-sys-dev/focal 0.8.0-1 arm64
mkchromecast-gstreamer/focal 0.3.8.1-1 all
phonon-backend-gstreamer-common/focal 4:4.10.0-1build1 arm64
phonon4qt5-backend-gstreamer/focal 4:4.10.0-1build1 arm64
qml-module-qtgstreamer/focal 1.2.0-5 arm64
qtgstreamer-doc/focal 1.2.0-5 all
qtgstreamer-plugins-qt5/focal 1.2.0-5 arm64
ruby-gstreamer/focal 3.4.1-2build1 arm64
ubuntu@ubuntu:~$

pallas, 2020-12-08:
Which GL are you using? On my Pi3B+ at least, X11 with Legacy didn't start, Full KMS didn't render anything from OMX, and Fake KMS did. IIRC, Pi4 only really supports Fake KMS but I wanted to confirm.

FD-, 2020-12-09:
A statement from a developer from Raspberry Pi Towers I found [here](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/98): 

> OpenMax IL will NEVER be supported on 64bit. The hell involved in mapping 64bit pointers to pass to the 32bit VPU just isn't worth it.

SumOlgy, 2020-12-09:
I’m playing around with usercfg.txt - here’s the currrent:

 this file. Please refer to the README file for a description of the various                                                                                      # configuration files on the boot partition.                                                                                                                       dtoverlay=dwc2,dr_mode=peripheral                                                                                                                                  dtoverlay=vc4-fkms-v3d                                                                                                                                             max_framebuffers=2                                                                                                                                                 gpu_mem=128                                                                                                                                                        hdmi_enable_4kp60=1                                                                                                                                                hdmi_ignore_edid=0xa5000080

From what FD said, am I right in thinking this will never work on a 64bit Pi OS?

pallas, 2020-12-19:
Yeah, it sounds like OMX will never work. However, gstreamer should.. I just don't know the incantation.

closed 2020-12-19 by pallas