FD-/RPiPlay #169: building on archlinuxarm

boschkundendienst, 2020-10-07:
Can someone try to build this on archlinuxarm. I think there are only some minor tweaks or correct environment variables necessary. Any help appreciated. Thanks in advance.

```
[root@dsclient build]# cmake ..
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenSSL: /usr/lib/libcrypto.so (found version "1.1.1h")
-- Found OpenMAX libraries for Raspberry Pi
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3")
-- Checking for modules 'gstreamer-1.0>=1.4;gstreamer-sdp-1.0>=1.4;gstreamer-video-1.0>=1.4;gstreamer-app-1.0>=1.4'
--   Found gstreamer-1.0, version 1.18.0
--   Found gstreamer-sdp-1.0, version 1.18.0
--   Found gstreamer-video-1.0, version 1.18.0
--   Found gstreamer-app-1.0, version 1.18.0
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBPLIST
    linked by target "airplay" in directory /root/RPiPlay/lib

CMake Error at renderers/CMakeLists.txt:52 (add_library):
  No SOURCES given to target: ilclient

CMake Generate step failed.  Build files cannot be regenerated correctly.
```

If I override this error I also get one for ilclient.

krzemin, 2020-10-11:
In `lib/CMakeLists.txt` try to replace:
```
find_library( LIBPLIST plist )
```
with 
```
find_library( LIBPLIST plist-2.0 )
```
Make sure that you have `libplist` package installed.

That helped to build rpiplay on my archlinuxarm.

FD-, 2020-10-11:
The problem rather seems to be caused by the lack of the ilclient library on distributions other than Raspberry Pi OS.

FD-, 2020-10-11:
That's why on Raspberry Pi, only Raspberry Pi OS is officially supported.

closed 2020-10-11 by FD-