FD-/RPiPlay #170: Overscan settings on tv

robstr98, 2020-10-09:
I'm running the raspberry on my tv an I needed to set overscan for it in /boot/config.txt.

When I use airplay with rpiplay these settings are ignored und the video is cut off at the edges.

Is there a way to set overscan in rpiplay?
Or do I have to set overscan for raspbian in another way?

pallas, 2020-10-09:
Are you sure you don't have to fix the overscan on your TV? I had a ton of problems that I assumed were due to RPi settings, but it turned out to be the TV. Once I tagged this particular HDMI port as a PC, the overscan issue disappeared.

robstr98, 2020-10-11:
Yes I'm sure. It was a relative cheap tv so apparently nobody thought of settings for overscan ...
I first thought exactly the same like you, so I looked it up already.

I have no problems except with rpiplay. The normal desktop fits perfect on the screen with the overscan settings enabled in the config.txt.
I have a FireTV stick and that has Overscan settings, too.