FD-/RPiPlay #179: Avahi issues

mat1987fer, 2020-11-02:
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>

Any ideas?

plm, 2020-11-02:
I looked into getting rid of this message and found it comes from [`avahi_warn_linkage` in `avahi-compat-libdns_sd/warn.c`](https://github.com/lathiat/avahi/blob/master/avahi-compat-libdns_sd/warn.c#L105). I believe RPiPlay links against `libdns_sd` to avoid needing separate dependencies and code paths for Linux and macOS builds.

I was able to disable this message by setting the `AVAHI_COMPAT_NOWARN` environment variable when starting `rpiplay`.

```shell
AVAHI_COMPAT_NOWARN=y ./rpiplay
```

Alternately, you can export the variable before starting RPiPlay.

```shell
export AVAHI_COMPAT_NOWARN=y
./rpiplay
```

pallas, 2020-11-02:
Yes, this is a duplicate of #51 and is a non-issue.

closed 2020-11-03 by mat1987fer