FD-/RPiPlay #180: Feedback from a RPiPlay newbie

djnugent, 2020-11-02:
### Be wary of package upgrades:
With 3B+, fresh install of `2020-08-20-raspios-buster-armhf-full.img`, and upgraded packages, `-r 90` and `-r 270` don't work. It rotates the image but the image is corrupted(jitter and fragmentation). This only happens if you use a fresh image AND upgrade packages during the initial setup process. If you refuse the upgrade, then there are no issues.  As a newbie, when the raspi setup program asks you if you want to upgrade out of date packages, it seems like a good idea. However, that is not the case here. 

### Gstreamer support
It appears that in order to enable gstreamer support, gstreamer needs be installed before you build RPi Play. Is that correct? I tried to install gstreamer after I built RPiPlay and it didn't work. When I figured out it was being linked at compile time, I made sure gstreamer was installed before building. This helped it find gstreamer at runtime, but I still had no success using gstreamer as a video renderer on the Pi. I think more docs need to be added for gstreamer support. Like is it even supported on Pi? Or is it just meant for desktop use?

### Rotated Display
I rotated the display (on the OS level) because I was having rotation issues as mentioned above. But an OS level rotation caused NULL buffer crashes. The behavior was `display_rotate=0` and `-r 0` works. But `display_rotate=1` and `-r 0` causes lag and NULL buffers. Note: this was running with upgraded packages. I did not replicate this test on a truly fresh install. 

Anyway, I finally got the system running stable, and I am not sure what caused a lot of these issues, but I think they can be avoided with more documentation.

FD-, 2020-11-02:
1. I wasn't aware of that issue! Sounds like it could be a regression in the ilclient or OpenMAX library (though I thought these were no longer actively developed) that should be reported to RPi Trading!
2. You're right. Please note though that at this point, Gstreamer doesn't really provide any benefit on a Raspberry Pi. I think gstreamer video rendering only works if you're using X11, and even then it's not accelerated.
3. Interesting! Could be related to 1. above.

djnugent, 2020-11-13:
I was running a 3B+. I am going to close this issue as there is nothing else to report or a proposed fix.

closed 2020-11-13 by djnugent