FD-/RPiPlay #181: Streaming udio only gives aacDecoder_DecodeFrame error

formatc1702, 2020-11-02:
Running RPiPlay with: 
```
./rpiplay -n PiPlay -a analog -vr dummy
```

Result:
- Regular AirPlay streaming (audio+video) works.
- Attempting to stream only audio (as per [this guide](https://www.cnet.com/how-to/airplay-only-the-audio-from-your-mac-using-os-x-mountain-lion/)) leads to a stream of errors:
  ```
  Accepted IPv4 client on socket 15
  Local: ...
  Remote: ...
  Accepted IPv4 client on socket 17
  Local: ...
  Remote: ...
  Found an unknown parameter: volume
  raop_rtp starting audio
  Got metadata of 143 bytes
  aacDecoder_DecodeFrame error : 0x4002
  aacDecoder_DecodeFrame error : 0x400c
  aacDecoder_DecodeFrame error : 0x4002
  aacDecoder_DecodeFrame error : 0x4002
  aacDecoder_DecodeFrame error : 0x4002
  aacDecoder_DecodeFrame error : 0x4002
  aacDecoder_DecodeFrame error : 0x400c
  aacDecoder_DecodeFrame error : 0x4002
  ...
  ```

For the last digit, I've seen `2`, `6` and `c`, no other values.

Setup:
- Raspberry Pi 1 (yes, don't have anything else to test now)
- Raspbian 10 (buster)
- RPiPlay 1.2

Am I doing something wrong? Does something need to be changed? Thanks!

pallas, 2020-11-02:
Thanks for the thorough bug report. Those error codes correspond to AAC_DEC_PARSE_ERROR, AAC_DEC_INVALID_CODE_BOOK, & AAC_DEC_TNS_READ_ERROR, which means the received bitstream is likely corrupt. I don't know if that's happening in the sender or in RPiPlay, but I can attempt to try this setup over the weekend, however I assume it's a duplicate of Issue #2. Apple uses ALAC for audio-only mirroring, while screen mirroring uses AAC.

formatc1702, 2020-11-03:
Thanks for the quick reply and interesting insight! I shall try Shairplay later today. If it turns out this *is* a duplicate of #2, let's close this issue. Will report back.

formatc1702, 2020-11-03:
Update: After messing around with Shairplay without success, I stumbled upon [shairport-sync](https://github.com/mikebrady/shairport-sync) that does just what I want, and works wonderfully. Thanks to the RPiPlay folks anyway! Have a good one.

closed 2020-11-03 by formatc1702