FD-/RPiPlay #192: CMake Error

angeloschl, 2020-12-06:
Hello,

after `cmake .. ` i get the following error. 

> CMake Error at /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
>   Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
>   system variable OPENSSL_ROOT_DIR: Found unsuitable version "1.1.0l", but
>   required is at least "1.1.1" (found
>   /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so)
> Call Stack (most recent call first):
>   /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:376 (_FPHSA_FAILURE_MESSAGE)
>   /usr/share/cmake-3.7/Modules/FindOpenSSL.cmake:375 (find_package_handle_standard_args)
>   lib/CMakeLists.txt:21 (find_package)

Any idea what to do?

Thanks!

pallas, 2020-12-06:
As the error says, you need a newer version of OpenSSL installed on the system.

angeloschl, 2020-12-06:
Thanks for you Help.
I tried `sudo apt-get install libssl-dev` multiple times but it still wont work. Is there an other way to get the newer version? 

I also did` sudo apt update sudo apt upgrade`

pallas, 2020-12-06:
I'm assuming that typing `openssl version` returns `1.1.0l`, but I have no idea what OS you're running. Is it RaspberryPi OS? If so, is it Wheezy, Jessie, Stretch, or Buster? If not, you need to check with your distribution.

angeloschl, 2020-12-06:
Yes it returns `1.1.0l`.
I am running Raspbian GNU/Linux 9 (stretch).

pallas, 2020-12-06:
Ug, OK, I don't see anything that actually depends on 1.1.0 features. This should have been set to the earliest working version with OPENSSL_API_COMPAT set. I'll reopen & fix.

pallas, 2020-12-07:
Can you please try with the latest commits?

angeloschl, 2020-12-07:
`cmake` worked!

> Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found suitable version "1.1.0l", minimum required is "1.1.0") 

But when i run the `make`i get this error now:

> lib/CMakeFiles/airplay.dir/build.make:86: die Regel für Ziel „lib/CMakeFiles/airplay.dir/crypto.c.o“ scheiterte
> make[2]: *** [lib/CMakeFiles/airplay.dir/crypto.c.o] Fehler 1
> CMakeFiles/Makefile2:239: die Regel für Ziel „lib/CMakeFiles/airplay.dir/all“ scheiterte
> make[1]: *** [lib/CMakeFiles/airplay.dir/all] Fehler 2
> Makefile:127: die Regel für Ziel „all“ scheiterte
> make: *** [all] Fehler 2

pallas, 2020-12-07:
Mein Deutsch spricht sehr schlecht. RPiPlay might actually depend on something in 1.1.1, in which case you'd need to upgrade to Buster. The make output above doesn't describe what the error was.. is there another log file? Can you provide the compiler output? I can't debug this any more today, my I might be able to take a look next week. @jasLogic do you recall off-hand whether any of the x25519 stuff depends on version 1.1.1 of OpenSSL?

pallas, 2020-12-07:
Ok, I lied, I did end up looking into this tonight. All of the x25519 stuff became first-class EVP in OpenSSL 1.1.1, with EVP_PKEY_X25519. Because the OpenSSL project considers 1.1.1 the stable branch and everything else deprecated, I don't think it's worth the energy to do compat for 1.0.2, although I'd consider patches that do so. In the mean time, I'm going to fix what I pushed earlier to fix this. I still want to keep their API compat flag for future people using newer OpenSSL.

tld;r: RPiPlay will not work prior to Buster any more. You need to upgrade your OS, as per https://www.raspberrypi.org/blog/buster-the-new-version-of-raspbian/

closed 2020-12-07 by pallas