FD-/RPiPlay #193: Segmentation Fault

mielune, 2020-12-09:
Hi,

I'm testing RPiPlay on Ubuntu 20.04 on a Razer Laptop (nvidia 1070 & intel GPU).

Client is an iPhone 11 with iOS 14.2, trying screen mirroring.

Just after the connection step, a coredump is raised.

No coredump file found.
Here the logs with debug activated (I tryed with and without -l option):

```
./rpiplay -l -d         
Using low-latency mode
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 21
Local: 192.168.68.107
Remote: 192.168.68.126
httpd receiving on socket 21
conn_request
Handling request GET with URL /info
INFO len = -1895839668
httpd receiving on socket 21
conn_request
Handling request POST with URL /pair-setup
httpd receiving on socket 21
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 21
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 21
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 21
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 21
conn_request
Handling request SETUP with URL rtsp://192.168.68.107/1661136025806589272
DACP-ID: C1671877E33BD551
Active-Remote: 2457178172
Transport: null
SETUP 1
eiv_len = 16
ekey_len = 72
fairplay_decrypt ret = 0
timing_rport = 56035
raop_ntp parse remote ip = 192.168.68.126
raop_ntp starting time
raop_rtp parse remote ip = 192.168.68.126
raop_rtp_mirror parse remote ip = 192.168.68.126
eport = 34059, tport = 37177
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = -1606605395131088
Accepted IPv4 client on socket 26
Local: 192.168.68.107
Remote: 192.168.68.126
httpd receiving on socket 21
conn_request
Handling request GET with URL /info
INFO len = -1895839668
httpd receiving on socket 21
conn_request
Handling request GET_PARAMETER with URL rtsp://192.168.68.107/1661136025806589272
httpd receiving on socket 21
conn_request
Handling request RECORD with URL rtsp://192.168.68.107/1661136025806589272
raop_handler_record
httpd receiving on socket 21
conn_request
Handling request SET_PARAMETER with URL rtsp://192.168.68.107/1661136025806589272
httpd receiving on socket 21
conn_request
Handling request SETUP with URL rtsp://192.168.68.107/1661136025806589272
DACP-ID: C1671877E33BD551
Active-Remote: 2457178172
Transport: null
type = 110
streamConnectionID = 17611729437630091219
raop_rtp_mirror starting mirroring
Mirroring initialized successfully
raop_rtp_mirror accepting client
raop_rtp_mirror width_source = 450.000000 height_source = 972.000000 width = 470.000000 height = 1016.000000
raop_rtp_mirror sps size = 17
raop_rtp_mirror pps size = 4
raop_rtp_mirror video ntp = 1607549474863204, now = 1607549474818180, latency = -45024
raop_rtp_mirror video ntp = 1607549474913202, now = 1607549474833210, latency = -79992
raop_rtp_mirror video ntp = 1607549474946533, now = 1607549474866732, latency = -79801
raop_rtp_mirror video ntp = 1607549474963199, now = 1607549474883677, latency = -79522
raop_rtp_mirror video ntp = 1607549474979865, now = 1607549474904724, latency = -75141
raop_rtp_mirror video ntp = 1607549474996531, now = 1607549474918010, latency = -78521
raop_rtp_mirror video ntp = 1607549475013197, now = 1607549474936837, latency = -76360
raop_rtp_mirror video ntp = 1607549475029863, now = 1607549474955172, latency = -74691
raop_rtp_mirror video ntp = 1607549475046529, now = 1607549474968889, latency = -77640
raop_rtp_mirror video ntp = 1607549475063194, now = 1607549474986694, latency = -76500
raop_rtp_mirror video ntp = 1607549475079860, now = 1607549475002675, latency = -77185
[1]    42399 segmentation fault (core dumped)  ./rpiplay -l -d
```

pallas, 2020-12-19:
Can you please provide a core dump or a backtrace?

mielune, 2020-12-19:
Where can I found the core dump on Ubuntu 20.04 ?

pallas, 2020-12-22:
Rebuild rpiplay with `-DCMAKE_BUILD_TYPE=Debug` (you have to rm CMakeCache.txt to do this)
Make sure you have gdb installed. 
Download this script: https://gist.github.com/pallas/035bf6beb9bbf0872d84b38ebacb0cde/raw/803ede36b919e460a863c7f913ec932c27ed44a5/gdb-bt.sh
Make it executable with `chmod +x gdb-bt.sh`
Run your regular command line with gdb-bt.sh in front of it, e.g. `./gdb-bt.sh ./rpiplay ...`
If it crashes, you should get a backtrace.

mielune, 2020-12-27:
Rebuilt with Debug and attach you have the stacktrace. Run with -d param.

[rpiplay.log](https://github.com/FD-/RPiPlay/files/5744422/rpiplay.log)

Hope it can help :smile:

pallas, 2020-12-29:
Thanks for the backtrace. That fault is deep inside gstreamer, and maybe inside your video driver. I'm pretty confident that it is not an rpiplay bug.

closed 2020-12-29 by pallas

mielune, 2020-12-29:
Ok

markuspi, 2021-08-31:
Getting the same error on Kubuntu 21.04. Backtrace of the crash points at the exact same location

[crash.log](https://github.com/FD-/RPiPlay/files/7084049/crash.log)

markuspi, 2021-08-31:
I followed [this comment](https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/285#note_689460)'s advice to install the non-free version of the `intel-media-va-driver` package, which solved my issue.

```bash
sudo apt install intel-media-va-driver-non-free
```