FD-/RPiPlay #195: No video pop-up for Ubuntu

SanderBorgmans, 2020-12-14:
Hi, I have been trying to cast my Ipad screen to my Ubuntu (18.04) laptop. While everything seems to work (the Ipad and laptop seem connected, using a wired connection for the laptop and the laptop hotspot for the ipad), no video output appears on my desktop after launching the rpiplay application and mirroring the screen of my Ipad. 

I get the following output (no debug flag):
```
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 29
Local: 10.42.0.1
Remote: 10.42.0.73
Accepted IPv4 client on socket 32
Local: 10.42.0.1
Remote: 10.42.0.73
raop_rtp_mirror starting mirroring
```

Using the debug flag, I get the following (stopped after a couple of seconds):
Initialized server socket(s)
```
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 30
Local: 10.42.0.1
Remote: 10.42.0.73
httpd receiving on socket 30
conn_request
Handling request GET with URL /info
INFO len = -352335604
httpd receiving on socket 30
conn_request
Handling request POST with URL /pair-setup
httpd receiving on socket 30
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 30
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 30
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 30
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 30
conn_request
Handling request SETUP with URL rtsp://10.42.0.1/6123499946918169206
DACP-ID: CC5A15E3531D28C
Active-Remote: 2183383873
Transport: null
SETUP 1
eiv_len = 16
ekey_len = 72
fairplay_decrypt ret = 0
timing_rport = 54875
raop_ntp parse remote ip = 10.42.0.73
raop_ntp starting time
raop_rtp parse remote ip = 10.42.0.73
raop_rtp_mirror parse remote ip = 10.42.0.73
eport = 33901, tport = 47970
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = -1607534488320372
Accepted IPv4 client on socket 32
Local: 10.42.0.1
Remote: 10.42.0.73
httpd receiving on socket 30
conn_request
Handling request GET with URL /info
INFO len = -352335604
httpd receiving on socket 30
conn_request
Handling request GET_PARAMETER with URL rtsp://10.42.0.1/6123499946918169206
httpd receiving on socket 30
conn_request
Handling request RECORD with URL rtsp://10.42.0.1/6123499946918169206
raop_handler_record
httpd receiving on socket 30
conn_request
Handling request SET_PARAMETER with URL rtsp://10.42.0.1/6123499946918169206
httpd receiving on socket 30
conn_request
Handling request SETUP with URL rtsp://10.42.0.1/6123499946918169206
DACP-ID: CC5A15E3531D28C
Active-Remote: 2183383873
Transport: null
type = 110
streamConnectionID = 12771718226008164549
raop_rtp_mirror starting mirroring
Mirroring initialized successfully
raop_rtp_mirror accepting client
raop_rtp_mirror width_source = 1298.000000 height_source = 972.000000 width = 1356.000000 height = 1016.000000
raop_rtp_mirror sps size = 18
raop_rtp_mirror pps size = 4
raop_rtp_mirror video ntp = 1607959310969878, now = 1607959310932882, latency = -36996
raop_rtp_mirror video ntp = 1607959310986545, now = 1607959310949448, latency = -37097
raop_rtp_mirror video ntp = 1607959311028246, now = 1607959310989910, latency = -38336
raop_rtp_mirror video ntp = 1607959311053260, now = 1607959311003263, latency = -49997
raop_rtp_mirror video ntp = 1607959311069937, now = 1607959311010759, latency = -59178
raop_rtp_mirror video ntp = 1607959311086613, now = 1607959311042623, latency = -43990
raop_rtp_mirror video ntp = 1607959311103290, now = 1607959311118994, latency = 15704
raop_rtp_mirror video ntp = 1607959311119966, now = 1607959311127074, latency = 7108
raop_rtp_mirror video ntp = 1607959311136643, now = 1607959311131533, latency = -5110
raop_rtp_mirror video ntp = 1607959311153319, now = 1607959311134519, latency = -18800
raop_rtp_mirror video ntp = 1607959311169996, now = 1607959311157738, latency = -12258
raop_rtp_mirror video ntp = 1607959311186672, now = 1607959311185372, latency = -1300
raop_rtp_mirror video ntp = 1607959311203349, now = 1607959311205788, latency = 2439
raop_rtp_mirror video ntp = 1607959311220025, now = 1607959311215235, latency = -4790
raop_rtp_mirror video ntp = 1607959311236702, now = 1607959311259574, latency = 22872
raop_rtp_mirror video ntp = 1607959311253378, now = 1607959311260921, latency = 7543
raop_rtp_mirror video ntp = 1607959311270055, now = 1607959311276151, latency = 6096
raop_rtp_mirror video ntp = 1607959311286731, now = 1607959311283144, latency = -3587
raop_rtp_mirror video ntp = 1607959311303408, now = 1607959311293587, latency = -9821
raop_rtp_mirror video ntp = 1607959311320084, now = 1607959311298163, latency = -21921
raop_rtp_mirror video ntp = 1607959311336761, now = 1607959311310896, latency = -25865
raop_rtp_mirror video ntp = 1607959311353437, now = 1607959311311213, latency = -42224
raop_rtp_mirror video ntp = 1607959311370104, now = 1607959311358804, latency = -11300
raop_rtp_mirror video ntp = 1607959311386771, now = 1607959311359118, latency = -27653
raop_rtp_mirror video ntp = 1607959311403437, now = 1607959311366434, latency = -37003
httpd receiving on socket 30
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
raop_rtp_mirror video ntp = 1607959312946043, now = 1607959312920272, latency = -25771
raop_rtp_mirror video ntp = 1607959312962710, now = 1607959312925294, latency = -37416
raop_rtp_mirror video ntp = 1607959312979376, now = 1607959312943491, latency = -35885
raop_rtp_mirror video ntp = 1607959312996043, now = 1607959312960809, latency = -35234
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = 0
httpd receiving on socket 30
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
raop_rtp_mirror video ntp = 1607959315164018, now = 1607959315106066, latency = -57952
raop_rtp_mirror video ntp = 1607959315214018, now = 1607959315217500, latency = 3482
raop_rtp_mirror video ntp = 1607959315297351, now = 1607959315296726, latency = -625
raop_rtp_mirror video ntp = 1607959315347351, now = 1607959315316348, latency = -31003
raop_rtp_mirror video ntp = 1607959316231364, now = 1607959316179309, latency = -52055
raop_rtp_mirror video ntp = 1607959316248030, now = 1607959316215421, latency = -32609
raop_rtp_mirror video ntp = 1607959316281364, now = 1607959316250988, latency = -30376
raop_rtp_mirror video ntp = 1607959316314697, now = 1607959316285484, latency = -29213
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = 4517
httpd receiving on socket 30
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
raop_rtp_mirror video ntp = 1607959317377525, now = 1607959317316730, latency = -60795
raop_rtp_mirror video ntp = 1607959317410859, now = 1607959317421305, latency = 10446
raop_rtp_mirror video ntp = 1607959317444192, now = 1607959317429511, latency = -14681
raop_rtp_mirror video ntp = 1607959317477525, now = 1607959317449629, latency = -27896
^CStopping...
httpd receiving on socket 30
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
Removing connection for socket 30
Destroying connection
raop_ntp stopping time thread
raop_ntp exiting thread
raop_ntp stopped time thread
raop_rtp_mirror exiting TCP thread
Removing connection for socket 32
Destroying connection
Exiting HTTP thread
```

Where is the video output going? Could I force it to show in some way?
Please let me know what info you require about my system(s). I should have installed the latest versions of the required packages using the installation documentation.

Thanks in advance.

screamingDanana, 2020-12-15:
I faced this problem too, but I solved it by allowing all incoming connection in ufw GUI. Restarted the server and screenshared my iPhone SE 1st Gen iOS 14.2. It should pop-up gstreamer player after a second.

I'm not sure of any other ways without disabling the firewall, though, unless there is an option to start the server at a specific port.

SanderBorgmans, 2020-12-15:
I should have enabled all connections. UFW is disabled and I used the following commands to open all ports for my hotspot connection (nm-shared is assigned as the firewall zone for the hotspot):

firewall-cmd --add-port=1025-65535/tcp --permanent --zone=nm-shared
firewall-cmd --add-port=1025-65535/udp --permanent --zone=nm-shared
firewall-cmd --reload

Should I be changing UFW settings as well?

pallas, 2020-12-19:
Have you seen PR#196?

pallas, 2020-12-19:
https://github.com/FD-/RPiPlay/pull/196

CO2020, 2021-01-11:
I have the same problem.
I am running Ubuntu 20.04 and even tried to use the static ports (#196).
Do you have any advice?
```
./rpiplay -d
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 28
Local: 192.168.179.160
Remote: 192.168.179.109
httpd receiving on socket 28
conn_request
Handling request GET with URL /info
INFO len = -1619011284
httpd receiving on socket 28
conn_request
Handling request POST with URL /pair-setup
httpd receiving on socket 28
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 28
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 28
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 28
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 28
conn_request
Handling request SETUP with URL rtsp://192.168.179.160/14296341740308023520
DACP-ID: 5365B482946A12E8
Active-Remote: 2174649013
Transport: null
SETUP 1
eiv_len = 16
ekey_len = 72
fairplay_decrypt ret = 0
timing_rport = 59051
raop_ntp parse remote ip = 192.168.179.109
raop_ntp starting time
raop_rtp parse remote ip = 192.168.179.109
raop_rtp_mirror parse remote ip = 192.168.179.109
eport = 7000, tport = 7011
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = -1610205404112555
Accepted IPv4 client on socket 30
Local: 192.168.179.160
Remote: 192.168.179.109
httpd receiving on socket 28
conn_request
Handling request GET with URL /info
INFO len = -1619011284
httpd receiving on socket 28
conn_request
Handling request GET_PARAMETER with URL rtsp://192.168.179.160/14296341740308023520
httpd receiving on socket 28
conn_request
Handling request RECORD with URL rtsp://192.168.179.160/14296341740308023520
raop_handler_record
httpd receiving on socket 28
conn_request
Handling request SET_PARAMETER with URL rtsp://192.168.179.160/14296341740308023520
httpd receiving on socket 28
conn_request
Handling request SETUP with URL rtsp://192.168.179.160/14296341740308023520
[... 530 lines cut ...]
Handling request POST with URL /feedback
raop_handler_feedback
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = 0
httpd receiving on socket 28
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = 0
httpd receiving on socket 28
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 28
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = 0
^Z
```

Waester, 2021-01-12:
@SanderBorgmans @CO2020 
I am no expert on the rpiplay application, but I know that when the initial connection is established and when the mirroring is shown on the laptop there should be at the minimum three network connections established (more if audio is being played).

Can you verify if that is the case? You should be able to run "**ss -tunp | grep rpiplay**" to see that.

If you see anything less than three, then there might still be something that is blocking.

SanderBorgmans, 2021-01-12:
@Waester I get the following output:

```
tcp   ESTAB  0       0                                         10.42.0.1:37785                       10.42.0.73:53930  users:(("rpiplay",pid=14255,fd=32))      
tcp   ESTAB  0       0                                         10.42.0.1:41293                       10.42.0.73:53931  users:(("rpiplay",pid=14255,fd=34))      
tcp   ESTAB  0       0                                         10.42.0.1:37785                       10.42.0.73:53929  users:(("rpiplay",pid=14255,fd=25)) 
```

And concerning the casting, I really want to use it for mirroring the whole screen. So that is not an issue.

CO2020, 2021-01-12:
@Waester I get the following output:
```
ss -tunp | grep rpiplay
tcp   ESTAB      0      0               192.168.179.160:7000    192.168.179.139:50293 users:(("rpiplay",pid=21836,fd=30))                                       
tcp   ESTAB      0      0               192.168.179.160:7100    192.168.179.139:50294 users:(("rpiplay",pid=21836,fd=32))                                       
tcp   ESTAB      0      0               192.168.179.160:7000    192.168.179.139:50292 users:(("rpiplay",pid=21836,fd=28))  
```
Any idea?
Thanks in advance.

Waester, 2021-01-12:
Network wise it seems to be fine for both of you.

Can you verify that all the dependacies is installed? If anything is missing you should install it and try again.

_sudo apt install libavahi-compat-libdnssd1,libplist3,libssl1.1,libgstreamer1.0-0,libgstreamer-plugins-base1.0-0,gstreamer1.0-libav,gstreamer1.0-vaapi,gstreamer1.0-plugins-bad_

Aside from that I have no further ideas, hopefully @FD-  or @pallas can help you troubleshoot further.

SanderBorgmans, 2021-01-12:
@Waester It seems fine, I get the following output:

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libplist3 is already the newest version (2.0.0-2ubuntu1).
libavahi-compat-libdnssd1 is already the newest version (0.7-3.1ubuntu1.2).
libavahi-compat-libdnssd1 set to manually installed.
libgstreamer-plugins-base1.0-0 is already the newest version (1.14.5-0ubuntu1~18.04.1).
libgstreamer1.0-0 is already the newest version (1.14.5-0ubuntu1~18.04.1).
libssl1.1 is already the newest version (1.1.1-1ubuntu2.1~18.04.7).
libssl1.1 set to manually installed.
gstreamer1.0-libav is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-plugins-bad is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-vaapi is already the newest version (1.14.5-0ubuntu1~ubuntu18.04.1).
0 t
```o upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.

CO2020, 2021-01-12:
@Waester It seems fine. I get the following output:

```
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
libavahi-compat-libdnssd1 ist schon die neueste Version (0.7-4ubuntu7).
libgstreamer-plugins-base1.0-0 ist schon die neueste Version (1.16.2-4).
libgstreamer1.0-0 ist schon die neueste Version (1.16.2-2).
libplist3 ist schon die neueste Version (2.1.0-4build2).
gstreamer1.0-libav ist schon die neueste Version (1.16.2-2).
gstreamer1.0-plugins-bad ist schon die neueste Version (1.16.2-2.1ubuntu1).
gstreamer1.0-vaapi ist schon die neueste Version (1.16.2-2).
libssl1.1 ist schon die neueste Version (1.1.1f-1ubuntu2.1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
```

CO2020, 2021-01-22:
I’m using Ubuntu 20.04 with nvidia drivers.

```
sudo apt remove gstreamer1.0-vaapi
sudo apt autoremove
```
- build it again
- it works!

Thanks!

_Originally posted by @fduncanh in https://github.com/antimof/UxPlay/issues/27#issuecomment-757442460_

SanderBorgmans, 2021-01-22:
@CO2020 Amazingly this works! Thank you so much.

tanrax, 2021-03-14:
That's how it worked for me. Thank you very much @CO2020

hdavid16, 2021-04-13:
Not working with Intel drivers

tanrax, 2021-04-13:
@hdavid16 I use Intel drives and it does work for me, what equipment do you have?

hdavid16, 2021-04-13:
> @hdavid16 I use Intel drives and it does work for me, what equipment do you have?

Intel 3rd gen core graphics
Intel core i7 processor
Ubuntu 20.04

kolaente, 2021-06-03:
I'm having the same issue on nixos (see #246). Additionally I see a lot of these errors:
```
** (rpiplay:12676): CRITICAL **: 13:29:31.348: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed
```

It seems to receiving video though, there's just no window opening to output it.

Full logs: https://gist.github.com/kolaente/e6afd18b07874fe56f2fe9a62b5e6b11

rogeriomcoelho, 2021-08-09:
> I’m using Ubuntu 20.04 with nvidia drivers.
> 
> ```
> sudo apt remove gstreamer1.0-vaapi
> sudo apt autoremove
> ```
> 
> * build it again
> * it works!
> 
> Thanks!
> 
> _Originally posted by @fduncanh in [antimof/UxPlay#27 (comment)](https://github.com/antimof/UxPlay/issues/27#issuecomment-757442460)_

Thanks man!!!! 
It's works for me!!!