FD-/RPiPlay #202: green screen in ubuntu

renau, 2020-12-19:
The default gstreamer generates a greenscreen (not working) in ubutun 20.10. The issue is the video decoder. When I apply this patch it works fine.

(I do not created a pull request because I am not familar with this code and it can break other things).

```
--- a/renderers/video_renderer_gstreamer.c
+++ b/renderers/video_renderer_gstreamer.c
@@ -69,7 +69,7 @@ video_renderer_t *video_renderer_gstreamer_init(logger_t *logger, video_renderer

     // Begin the video pipeline
     GString *launch = g_string_new("appsrc name=video_source stream-type=0 format=GST_FORMAT_TIME is-live=true !"
-                                   "queue ! decodebin ! videoconvert ! ");
+                                   "queue ! h264parse ! avdec_h264 ! ");
     // Setup rotation
     if (config->rotation != 0) {
         switch (config->rotation) {
```

An issue is that in ubuntu, this requires this package: gstreamer1.0-plugins-bad

pallas, 2020-12-19:
@dougg3, do you know?

dougg3, 2020-12-19:
Interesting...I can't reproduce the problem. I quickly tried downloading an Ubuntu 20.10 virtual machine and ran it in VMware with the stock RPiPlay. It works -- no green screen. (However, there's some kind of problem with the stride that causes the video display to be completely corrupted, which is a problem I've been seeing with RPiPlay + VMware for a long time that only happens inside of VMware...not on actual hardware)

The interesting thing is that applying the patch completely breaks the video display. A video window never pops up.

I did want to point out that the compile instructions already direct you to install gstreamer1.0-plugins-bad. I believe it's necessary in order to provide the H.264 decoding support.

marcoscannabrava, 2021-08-11:
@renau thank you! I had this same problem on PopOS 21.04 and this one-line patch worked like a charm.