FD-/RPiPlay #203: Video artifacts when mirroring from iOS device but not macOS device

dustinlieu, 2020-12-23:
I'm running RPiPlay on an Ubuntu VM. I get major video artifacts when mirroring from an iOS device but not from a macOS device. When I set `displays_0_overscanned_node` to `0` in lib/raop_handlers.h, it fixes the diagonal shifting effect, but there are still problems with the colors.

Default:
[image]

With overscan disabled:
[image]

dustinlieu, 2020-12-26:
When I change the resolution in `lib/raop_handlers.h` to 1280x720 it displays perfectly. Not sure why it only messes up at 1920x1080.

gusseroni, 2021-03-23:
I also encountered this, I had to both disable overscan and set the resolution to exactly that of the iOS device:

```
$ git diff
diff --git a/lib/raop_handlers.h b/lib/raop_handlers.h
index 104b1c4..7de4806 100755
--- a/lib/raop_handlers.h
+++ b/lib/raop_handlers.h
@@ -134,13 +134,13 @@ raop_handler_info(raop_conn_t *conn,
     plist_t displays_0_uuid_node = plist_new_string("e0ff8a27-6738-3d56-8a16-cc53aacee925");
     plist_t displays_0_width_physical_node = plist_new_uint(0);
     plist_t displays_0_height_physical_node = plist_new_uint(0);
-    plist_t displays_0_width_node = plist_new_uint(1920);
-    plist_t displays_0_height_node = plist_new_uint(1080);
-    plist_t displays_0_width_pixels_node = plist_new_uint(1920);
-    plist_t displays_0_height_pixels_node = plist_new_uint(1080);
+    plist_t displays_0_width_node = plist_new_uint(2048);
+    plist_t displays_0_height_node = plist_new_uint(1536);
+    plist_t displays_0_width_pixels_node = plist_new_uint(2048);
+    plist_t displays_0_height_pixels_node = plist_new_uint(1536);
     plist_t displays_0_rotation_node = plist_new_bool(0);
     plist_t displays_0_refresh_rate_node = plist_new_real(1.0 / 60.0);
-    plist_t displays_0_overscanned_node = plist_new_bool(1);
+    plist_t displays_0_overscanned_node = plist_new_bool(0);
     plist_t displays_0_features = plist_new_uint(14);
 
     plist_dict_set_item(displays_0_node, "uuid", displays_0_uuid_node);
```