FD-/RPiPlay #204: Closing Casting Window Does Not End Application On Fedora Linux

progressEdd, 2020-12-23:
When a user stops screen mirroring and closes the window on Linux, the service does not stop. If a user closes the window, and tries to restart airplay, the video casting won't show. The user will be confused as to why their audio is casted, but not the screen. 

The following video illustrates this issue
https://youtu.be/HHCIv0GU25I

A potential solution would have the RPiPlay service end upon window closure. Alternatively, the user can close the window and have the service continue, but if a new airplay cast is started, a new screen share window can be opened.

The test environment is
- Operating System: Fedora 33 (x86-64)
- Desktop Environment: Cinnamon 4.6.7
- Linux Kernel: 5.9.13-200.fc33.x86_64
- Processor: AMD Ryzen 7 2700X Eight-Core Processor × 8
- RAM: 16 GiB
- Graphics: NVIDIA Corporation TU106 [GeForce RTX 2070]
- Graphics Driver: NVIDIA Proprietary Driver 455.45.01

pallas, 2020-12-24:
Is this different than #82 ? Does the script referred to in my comment help?

progressEdd, 2020-12-24:
> Is this different than #82 ? Does the script referred to in my comment help?
The script is a good solution for this suggestion
> Alternatively, the user can close the window and have the service continue, but if a new airplay cast is started, a new screen share window can be opened.

It's slightly different because as long as the casting window remains open, I do not have the same issue as the OP in that thread. If I stop the cast and recast later it will continue to work. 

Really it depends on use case. 
If we want to make this a gnome shell extension, we would want the service to continue running in the background and have each cast open a new window. In the case of a raspberry pi casting solution, you'd probably want the service to continue running in the background so it can be on standby for any requests. This isn't as big of issue on raspberry pi's because the user won't have as much interaction with the casting window as a desktop user.

If we are making it a desktop application, your script would be better because stopping the cast would close the application and would prompt the user to open a new instance of the application for each casting session.

fast-potat0, 2021-03-30:
> the user can close the window and have the service continue, but if a new airplay cast is started, a new screen share window can be opened.
I think the best solution is when the window is destroyed when mirroring is ended and then initiate it again when necessary. @rbribeiro have you found where the window is initiated?

> Is this different than #82 ? Does the script referred to in my comment help?

Script from this issue didn't help me. It resized my chrome window instead of rpiplay window.