FD-/RPiPlay #208: build issues for macOS

lewisxy, 2021-01-09:
I tried to build this on macOS 10.14.6 and I installed all dependencies using `brew`
```
brew install cmake libplist nss gstreamer gst-plugins-bad gst-plugins-base
```
but the build failed. After a bit research, I fixed the issue by adding the following lines to top level CMakeList.txt
```
include_directories(/usr/local/include)
link_directories(/usr/local/lib)
```
and add the following lines to `lib/raop_rtp_mirror.c` after all the "#include"
```
#if !defined(SOL_TCP) && defined(IPPROTO_TCP)
#define SOL_TCP IPPROTO_TCP
#endif
#if !defined(TCP_KEEPIDLE) && defined(TCP_KEEPALIVE)
#define TCP_KEEPIDLE TCP_KEEPALIVE
#endif
```
I am not entirely sure of how to modify the makefile and c source code so that it can be build on both Linux and macOS, so I am not going to submit a PR for this. If anyone know how to do it, please fix this.

------------------------------

Now I am able to run the binary on macOS, but when I mirror the screen of my iPhone or iPad, nothing showed up (unlike on Linux, which opened a new window). I am not sure why this is the case. Are there any other ways to get the stream out?

pigworlds, 2021-02-27:
I hit the same issue. The missing window turn out to because of [this](https://stackoverflow.com/questions/35137165/gstreamer-1-0-video-from-tutorials-is-not-playing-on-macos).

You need to add Glib main event loop. I added it to the main() and does popup the window.

alyyousuf7, 2021-03-28:
https://github.com/FD-/RPiPlay/pull/155 -- This PR adds Glib main event loop