FD-/RPiPlay #209: Cannot play video

lewisxy, 2021-01-09:
The screen mirroring works fine along with audio. But when I try to play a video via airplay (for example, use airplay from youtube app), it does not work. The server did output some log suggesting a connection is made and even `raop_rtp starting audio`. But no video or audio is being outputted. I tested it from 2 devices (iOS 14.3, iOS 12) and the result is the same. I also tested some commercial product such as Reflector 3 using the same iOS devices, which does work (and even support video playback controls such as play/pause and drag). Is video feature currently not supported?

FD-, 2021-01-09:
Duplicate of #7

closed 2021-01-09 by FD-

lewisxy, 2021-01-10:
> AirPlay video (any video streaming that is not initiated via the dedicated AirPlay mirroring button in Control Center) uses a different protocol that has the iOS device relay the url and information of the (HLS) video it is currently playing to the AirPlay server. The server would then start streaming the video directly from the server (ie Youtube in this case), and the iOS device would act as a remote control for pausing, seeking and so on.

This is not the complete story. From the system design perspective, if the video is hosted in a remote server, there is good reason to let the receiving end (server) to stream it using the information provided by the sender (client). However, local videos also works. For example, open a video in Photos app on iOS device, click share button, and there is an "AirPlay" option to play it on AppleTV. The video is local, and it does not screen mirroring (the receiving end will play it as if the video is full screened, but on the device it's not).

lewisxy, 2021-01-10:
> For example, open a video in Photos app on iOS device, click share button, and there is an "AirPlay" option to play it on AppleTV. The video is local, and it does not screen mirroring (the receiving end will play it as if the video is full screened, but on the device it's not).

UPDATE: I just tested it. Despite the fact that this scenario is technically not screen mirroring, it does appear to use the same protocol, and I can get it to work (i.e. the video does play on this program as if it was mirroring). Based on this result, I suppose you are  right that any video stream generated locally (whether it's screen mirroring or not) uses the same protocol, whereas video in remote uses a different protocol.

If that is the case, I have no problem with the remote video case not working (since it's being streamed from the receiving end anyway), and thanks for the explanation. I am writing this down so that others won't need to ask the same question.

FD-, 2021-01-10:
Keep in mind a fullscreen video doesn't necessarily mean that the video protocol is involved. Apps can render custom UI to the secondary screen, which might be what the Photos app is doing.