FD-/RPiPlay #210: make: dns_sd.h: No such file or directory

drscotthawley, 2021-01-11:
Same issues as https://github.com/antimof/UxPlay/issues/30:  Make produces an error message about not being able to find dns_sh.h, on Ubuntu 20.10:

```
RPiPlay/lib/dnssd.c:38:10: fatal error: dns_sd.h: No such file or directory
   38 | #include <dns_sd.h>
```
A common fix for this seen around the internet is to install libavahi-compat-libdnssd-dev, however as this was already the latest version on my system, doing this had no effect. 
 

A fix I found for this was to manually download dns_sd.h from Apple, put it in RPIplay/lib, and change the angle-brackets to quotes in the include statement on line 38 of lib/dnssd.c 

After that, build occurs without errors.   However, one sees errors from gstreamer and the iPad, after finding RPiPlay, says "Unable to connect to RPIPlay"  

Log shows...

```
$ ./rpiplay 
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>

(rpiplay:652076): GStreamer-WARNING **: 12:55:59.586: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstlibav.so': /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end

(rpiplay:652076): GStreamer-WARNING **: 12:55:59.613: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstlibav.so': /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end

^CStopping...
```

drscotthawley, 2021-01-11:
Fixed the 'Failed to load plugin' errors by removing parts of LDFLAGS environment variable that were set by Anaconda, and did a clean rebuild.  Only warnings now are the generic Apple-Avahi warnings (the first 3 lines above).

'Still getting "Unable to connect to "RPiPlay"' from iPad.

drscotthawley, 2021-01-11:
After modifying code to use static ports as per https://github.com/FD-/RPiPlay/pull/196, rebuilding, and opening firewall, I now get a connection and then immediate segfault:

```
$ ./rpiplay 
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 24
Local: 10.0.1.8
Remote: 10.0.1.3
Accepted IPv4 client on socket 26
Local: 10.0.1.8
Remote: 10.0.1.3
raop_rtp_mirror starting mirroring
Segmentation fault (core dumped)
```

Will try to rebuild with debug and share a stack trace later.

drscotthawley, 2021-01-14:
After following the debug instructions in https://github.com/FD-/RPiPlay/issues/193, here's the output when it core dumps:

```
~/Downloads/RPiPlay/build$ ./gdb-bt.sh ./rpiplay 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff674b640 (LWP 1264010)]
[Thread 0x7ffff674b640 (LWP 1264010) exited]
[New Thread 0x7ffff674b640 (LWP 1264011)]
[New Thread 0x7ffff5f4a640 (LWP 1264012)]
[New Thread 0x7ffff5749640 (LWP 1264013)]
[New Thread 0x7ffff4a51640 (LWP 1264014)]
[Thread 0x7ffff4a51640 (LWP 1264014) exited]
[New Thread 0x7ffff4a51640 (LWP 1264015)]
[New Thread 0x7fffd3fff640 (LWP 1264016)]
[New Thread 0x7fffd37fe640 (LWP 1264017)]
Initialized server socket(s)
[New Thread 0x7fffd2ffd640 (LWP 1264018)]
[New Thread 0x7fffd27fc640 (LWP 1264019)]
[New Thread 0x7fffd1ffb640 (LWP 1264020)]
Accepted IPv4 client on socket 24
Local: 10.0.1.8
Remote: 10.0.1.2
[New Thread 0x7fffbd4c4640 (LWP 1264022)]
Accepted IPv4 client on socket 26
Local: 10.0.1.8
Remote: 10.0.1.2
raop_rtp_mirror starting mirroring
[New Thread 0x7fffbccc3640 (LWP 1264023)]
[New Thread 0x7fffb789c640 (LWP 1264024)]
[Thread 0x7fffb789c640 (LWP 1264024) exited]
[New Thread 0x7fffb789c640 (LWP 1264025)]
[New Thread 0x7fffb709b640 (LWP 1264026)]
[New Thread 0x7fffb5e9d640 (LWP 1264027)]

Thread 5 "queue0:src" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff5749640 (LWP 1264013)]
0x00007fffb7a8abc5 in ?? () from /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
#0  0x00007fffb7a8abc5 in  () at /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
#1  0x00007fffb7a7bbf4 in  () at /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
#2  0x00007fffb7a6261a in  () at /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
#3  0x00007fffb7c705e3 in  () at /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
#4  0x00007fffb7c74016 in  () at /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
#5  0x00007fffb7c7429a in  () at /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
#6  0x00007fffb7c41f92 in  () at /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
#7  0x00007fffd00532b0 in vaPutImage () at /lib/x86_64-linux-gnu/libva.so.2
#8  0x00007fffbc318a14 in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstvaapi.so
#9  0x00007fffbc2d69c6 in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstvaapi.so
#10 0x00007fffbc2ddcce in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstvaapi.so
#11 0x00007ffff741cc33 in  () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#12 0x00007ffff742064d in  () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#13 0x00007ffff7ec7208 in gst_pad_query () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#14 0x00007ffff7ec796b in gst_pad_peer_query () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#15 0x00007ffff7f073c8 in gst_pad_peer_query_caps () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#16 0x00007ffff742032f in  () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#17 0x00007ffff7ec7208 in gst_pad_query () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#18 0x00007ffff7ec796b in gst_pad_peer_query () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#19 0x00007ffff7f015fc in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#20 0x00007ffff7ec5b5e in gst_pad_forward () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#21 0x00007ffff7f0429a in gst_pad_proxy_query_caps () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#22 0x00007ffff7ec5d88 in gst_pad_query_default () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#23 0x00007ffff7ec7208 in gst_pad_query () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#24 0x00007ffff7ec796b in gst_pad_peer_query () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#25 0x00007ffff7f073c8 in gst_pad_peer_query_caps () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#26 0x00007ffff6c76631 in  () at /lib/x86_64-linux-gnu/libgstvideo-1.0.so.0
#27 0x00007fffbc2d04f8 in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstvaapi.so
#28 0x00007ffff6c64a66 in  () at /lib/x86_64-linux-gnu/libgstvideo-1.0.so.0
#29 0x00007ffff7ec7208 in gst_pad_query () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#30 0x00007ffff7f04b38 in gst_pad_query_caps () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#31 0x00007ffff6c64cd1 in  () at /lib/x86_64-linux-gnu/libgstvideo-1.0.so.0
[... 34 lines cut ...]
#66 0x00007ffff7ec9530 in gst_pad_push_event () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#67 0x00007fffbc47bef0 in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstvideoparsersbad.so
#68 0x00007fffbc47df9e in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstvideoparsersbad.so
#69 0x00007fffbc4802da in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstvideoparsersbad.so
#70 0x00007ffff73f7620 in  () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#71 0x00007ffff73fd37e in  () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#72 0x00007ffff7ebec0d in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#73 0x00007ffff7ec0e89 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#74 0x00007ffff7ec8492 in gst_pad_push () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#75 0x00007ffff7ebec0d in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#76 0x00007ffff7ec0e89 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#77 0x00007ffff7ec8492 in gst_pad_push () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#78 0x00007ffff7eabbf3 in gst_proxy_pad_chain_default () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
[... cut ...]
#82 0x00007ffff6e6cdb3 in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so
#83 0x00007ffff7ef81c7 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#84 0x00007ffff7d27ac4 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#85 0x00007ffff7d271b1 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#86 0x00007ffff7c91590 in start_thread (arg=0x7ffff5749640) at pthread_create.c:463
#87 0x00007ffff756f223 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

Any insights would be appreciated.

freiheitsnetz, 2021-03-01:
Hi. Don'know whether it's related but in some other case the segmentation fault was related to gstreamer1.0-vaapi.
https://github.com/FD-/RPiPlay/issues/195#issuecomment-765556823
https://github.com/antimof/UxPlay/issues/32
Could be a try to remove that one.

kod3000, 2021-07-19:
I was having the same problem "segmentation fault (core dumped)", with 3 different graphics modes.. Using the solution provided by @freiheitsnetz and rebuilding rpiplay the problem was solved. 👍🏽

ryota2425, 2021-09-17:
```
sudo apt-get install libavahi-compat-libdnssd-dev
```

this helps me!