FD-/RPiPlay #211: GStreamer not recognized (Arch Linux)

ThatFruedDued, 2021-01-11:
Hello,
When I run RPiPlay, GStreamer does not seem to be recognized. I am running Arch Linux and I have verified that GStreamer is installed and working. 
[image]
All of the other features of the app work. I can connect to RPiPlay from devices, I just cannot view the display of the device or listen to the audio (although audio support is not a necessity for me).
How can I fix this, or is this simply incompatibility with Arch Linux? Thanks!

SyzSteM, 2021-01-12:
If you want, you can take a look at my AUR package of this project here: [rpiplay-git](https://aur.archlinux.org/packages/rpiplay-git/).
Maybe you are missing the gstreamer bad plugins (gst-plugins-bad)? You could try to install my package from the AUR, it tested it on clean arch.
Cheers!

ThatFruedDued, 2021-01-12:
@SyzSteM I'll look into yours! I have gst-plugins-bad installed, so I'm unsure why it's not working.

ThatFruedDued, 2021-01-13:
@SyzSteM no luck. Could it have to do with conflicts with KDE Plasma? Thanks for your help!

ThatFruedDued, 2021-01-13:
I got it! I used your AUR package, and now it displays on my laptop. Thanks!

closed 2021-01-13 by ThatFruedDued