FD-/RPiPlay #216: audio is choppy

nr2003, 2021-01-23:
Can someone fix the audio? I have reviewed other posts on the audio but it isn't working for me and in the -l flagged mode It is delayed big time. I am using a pi 3B. What can I do?

bpintea, 2021-02-14:
1st, thanks a lot for making this available! Then, sorry for posting here, if there's a forum, would happily use that.
I have quasi the same issue: running a fresh Release build (5563bf4c8d) on a RPi 4B 4GB on Raspbian, 256 MB video memory, everything ethernet and having two different choppy experiences: 
* with `-l` left out, the video is smooth, but the audio choppy, over both analog and HDMI outputs (at a time); no noticeable delay, though.
* with `-l`, the video is choppy (not a lot, but still noticeable), the audio is normal (smooth).

Running with lowered niceness or realtime priority makes no difference.

Any pointers appreciated. Cheers.

smaktacular, 2021-04-15:
Same here, on a 3B+. Also wired connection.

philffm, 2021-05-04:
Same behaviour here - RPi 3B+ 

Yet still remarkable, how good it works!