FD-/RPiPlay #218: startup casting window in fullscreen

bendschs, 2021-01-26:
i found a way to maximize the window by cli:
wmctrl -r :ACTIVE: -b toggle,fullscreen

the problem now is, that the Rpiplay-window is created only when something is being casted to rpiplay. I am looking for a way to tell the window being casted to immediately and automatically switch to full-screen.

any help

srbhp, 2021-03-03:
`wmctrl -r rpiplay -b toggle,fullscreen`  works for me after starting rpiplay. 
May be this is the same thing what you have done. but your command didn't worked for me.

Loremwriter, 2022-04-30:
Hi,
i installed wmctrl and tried your code but it doesnt work for me. 
How do i have to use your code?
It just tells me "Cannot open display". 

Would be glad to get any help to start the casting screen in fullscreen at boot. 
Can anyone help me? :)

bendschs, 2022-05-02:
i was successful with the following commands:

```
export DISPLAY=:0.0
wmctrl -r :ACTIVE: -b toggle,fullscreen
```

cheers
bendsch