FD-/RPiPlay #219: No Audio

FayardProf, 2021-02-01:
Hi,

I have a Raspberry Pi 4 B with the latest Raspberry Pi Os (the Desktop 32-bit version). I have removed pulseaudio as I had many problems with it (OS freezing). Now, the sound is working ok on my system with VLC (I am using the headphones to check it). But when I am using RPiPlay, there seems to have no error on the iPad I am streaming from, but I get no audio on the Raspberry Pi. The video is fine thought.

What can I do to check if the audio is received correclty on the Pi?

Best regards,
François

JaegerP, 2021-02-03:
Having the same issue on a desktop Fedora 32 install (64bit). I tried running`GST_DEBUG=3 ./rpiplay`.

It seems to set up the connection correctly, but apparently libav can't decode the streaam:
```
Accepted IPv4 client on socket 28
Local: 192.168.1.100
Remote: 192.168.1.102
Accepted IPv4 client on socket 30
Local: 192.168.1.100
Remote: 192.168.1.102
raop_rtp starting audio
0:00:08.817086672 319748       0xb96b00 ERROR                  libav :0:: Number of bands (16) exceeds limit (8).
0:00:08.817110269 319748       0xb96b00 WARN                   libav gstavauddec.c:756:gst_ffmpegauddec_handle_frame:<avdec_aac0> decoding error
0:00:08.817211846 319748       0xb96b00 ERROR                  libav :0:: Number of bands (16) exceeds limit (8).
0:00:08.817228974 319748       0xb96b00 WARN                   libav gstavauddec.c:756:gst_ffmpegauddec_handle_frame:<avdec_aac0> decoding error
0:00:08.835504754 319748       0xb96b00 ERROR                  libav :0:: Number of bands (16) exceeds limit (8).
0:00:08.835529028 319748       0xb96b00 WARN                   libav gstavauddec.c:756:gst_ffmpegauddec_handle_frame:<avdec_aac0> decoding error
0:00:08.835546805 319748       0xb96b00 ERROR                  libav :0:: Number of bands (16) exceeds limit (8).
```

I have no experience with the inner workings of gstreamer and/or libav but maybe there is a way to increase the band limit?

roobre, 2021-02-08:
Same issue here on Arch Linux:

```
[roobre@Gigabox build]$ ./rpiplay -vr gstreamer -ar gstreamer -n Gigabox
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 25
Local: 10.0.0.5
Remote: 10.0.0.201
Accepted IPv4 client on socket 28
Local: 10.0.0.5
Remote: 10.0.0.201
raop_rtp starting audio
raop_rtp starting audio
raop_rtp starting audio
^CStopping...
Removing connection for socket 25
Destroying connection
Removing connection for socket 28
Destroying connection
[roobre@Gigabox build]$ GST_DEBUG=3 ./rpiplay -vr gstreamer -ar gstreamer -n Gigabox
0:00:00.103355470 268203 0x5650a28c2730 WARN                gleglgbm gstgl_gbm_utils.c:489:gst_gl_gbm_find_and_open_drm_node: Cannot open device node "/dev/dri/card0": Permission denied (13)
0:00:00.104039560 268203 0x5650a28c2730 ERROR              gldisplay gstgldisplay_gbm.c:439:gst_gl_display_gbm_new: could not find or open DRM device
0:00:00.109061235 268203 0x5650a28c2730 WARN                glwindow gstglwindow.c:323:gst_gl_window_new: Could not create window. user specified (null), creating dummy window
0:00:00.109830687 268203 0x7f5e5c0020c0 WARN               glcontext gstglcontext.c:1257:gst_gl_context_create_thread:<glcontextegl0> Failed to create context
0:00:00.110103738 268203 0x5650a28c2730 WARN             glimagesink gstglimagesink.c:1095:_ensure_gl_setup:<sink> error: Failed to initialize egl: EGL_NOT_INITIALIZED
0:00:00.116977918 268203 0x5650a28c2730 WARN                 kmssink gstkmssink.c:848:gst_kms_sink_start:<video_sink-actual-sink-kms> error: Could not open DRM module (NULL)
0:00:00.117247707 268203 0x5650a28c2730 WARN                 kmssink gstkmssink.c:848:gst_kms_sink_start:<video_sink-actual-sink-kms> error: reason: No such file or directory (2)
0:00:00.117464332 268203 0x5650a28c2730 WARN                basesink gstbasesink.c:5865:gst_base_sink_change_state:<video_sink-actual-sink-kms> error: Failed to start
0:00:00.127284720 268203 0x5650a28c2730 WARN                   vaapi gstvaapiutils.c:78:gst_vaapi_warning: /usr/lib/dri/iHD_drv_video.so init failed
0:00:00.149790829 268203 0x5650a28c2730 WARN                   vaapi gstvaapiutils.c:78:gst_vaapi_warning: /usr/lib/dri/iHD_drv_video.so init failed
0:00:00.152203073 268203 0x5650a28c2730 WARN              vaapiblend gstvaapiblend.c:184:gst_vaapi_blend_initialize:<vaapiblend0> VPP does not support global alpha blending
0:00:00.184652758 268203 0x5650a28c2730 WARN                   vaapi gstvaapiutils.c:78:gst_vaapi_warning: /usr/lib/dri/iHD_drv_video.so init failed
0:00:00.217040042 268203 0x5650a28c2730 WARN                   vaapi gstvaapiutils.c:78:gst_vaapi_warning: /usr/lib/dri/iHD_drv_video.so init failed
0:00:00.230753791 268203 0x5650a2879860 FIXME                default gstutils.c:4025:gst_pad_create_stream_id_internal:<video_source:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Initialized server socket(s)
0:00:00.251431220 268203 0x7f5e5c002120 FIXME                default gstutils.c:4025:gst_pad_create_stream_id_internal:<audio_source:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 26
Local: 10.0.0.5
Remote: 10.0.0.201
Accepted IPv4 client on socket 28
Local: 10.0.0.5
Remote: 10.0.0.201
raop_rtp starting audio
raop_rtp starting audio
0:00:02.864824325 268203 0x7f5e5c002000 ERROR                  libav :0:: Number of bands (16) exceeds limit (8).
0:00:02.864990750 268203 0x7f5e5c002000 WARN                   libav gstavauddec.c:774:gst_ffmpegauddec_handle_frame:<avdec_aac0> decoding error
0:00:02.895001466 268203 0x7f5e5c002000 ERROR                  libav :0:: Number of bands (16) exceeds limit (8).
0:00:02.895102817 268203 0x7f5e5c002000 WARN                   libav gstavauddec.c:774:gst_ffmpegauddec_handle_frame:<avdec_aac0> decoding error
0:00:02.895223417 268203 0x7f5e5c002000 ERROR                  libav :0:: Number of bands (16) exceeds limit (8).
0:00:02.895269530 268203 0x7f5e5c002000 WARN                   libav gstavauddec.c:774:gst_ffmpegauddec_handle_frame:<avdec_aac0> decoding error
0:00:02.895419017 268203 0x7f5e5c002000 ERROR                  libav :0:: Number of bands (16) exceeds limit (8).
0:00:02.895888894 268203 0x7f5e5c002000 WARN                   libav gstavauddec.c:774:gst_ffmpegauddec_handle_frame:<avdec_aac0> decoding error
[ two lines above repeat forever ]
```

Seems like gstreamer is not liking some of the parameters set by Rpiplay, but I don't have the background needed to know more.
If there's anything I can test to help, please just @ me.

JaegerP, 2021-02-08:
> EDIT: Seems like https://github.com/FD-/RPiPlay/issues/2 is to blame, at least for me (I'm trying to stream audio only).

shairplay (https://github.com/juhovh/shairplay) works for me as well. I didn't have any chance yet to see if some code snippets could be merged here - would be nice to be able to share video as well. If there is some interested I can look into it.

roobre, 2021-02-09:
I looked into Shairplay, but it seems to be using Airplay 1, which enforces a 2-second delay to all audio transmission. My use case involves teleconferencing, so it is a deal breaker for me :(

It was my understanding that this project used Airplay 2, which didn't have such limitaiton. However it seems like it's going to be complicated to implement audio-only receving without deeper knowledge of the protocol.