FD-/RPiPlay #221: vnc doesn't transmit AirPlay mirrored video

ivancich, 2021-02-16:
I have a relative in their late 80s who lives 2000+ miles from me. They're not technically inclined. They have an iPad and remote support has been incredibly difficult. What's on the screen? What are your options?

So I thought I could send them a Raspberry Pi with this software plus RealVNC (realvnc.com) on it. Have them plug it into their router. Then, when they need support, they could mirror to the Pi and I'd be able to see their screen through VNC.

I just tried this out, and while the AirPlay mirroring appears on the HDMI port, it is not copied over to the VNC connection.

I'm not claiming this is a bug or an issue. At this point I'm wondering if anyone knows a) why this might be, b) whether there might be a fix, and c) how complex the fix would be.

I haven't explored the source code yet, but I'm well versed in C/C++ and perhaps with some direction I could lend a hand.

pallas, 2021-02-17:
You might try the gstreamer version. The original version writes directly into the GPU, which is why VNC never sees it. But why not use Apple's built in "share my screen with my friend" feature?

ivancich, 2021-02-17:
Thank you for the insights, @pallas. I was able to install the various gstreamer components and rebuild. And I did get the following command-line to work successfully:

`rpiplay -vr gstreamer | gst-launch-1.0 -v fdsrc`

The video stream was shared via VNC.

Now I'm completely new to gstreamer, and the learning curve looks pretty steep. Is that command above the best way to get things going?

Also, I've been looking for a while for a way so that my relative can share his screen. Some apps claim to do this by leveraging iOS's screen-recording feature, but when I've tried those, they fail early and often with errors that seem to relate to DRM (i.e., detecting drm'd content and stopping).

So can you say more about the "share my screen with my friend" under iOS? Is it through an app? Through the os?

But thank you! Whatever else happens, I believe I have a workable solution....

FD-, 2021-02-18:
Actually, it is possible to use the default Raspberry PI renderer with VNC. You have to enable the "Experimental Direct Capture Mode" in RealVNC: https://help.realvnc.com/hc/en-us/articles/360002249917-VNC-Connect-and-Raspberry-Pi#running-directly-rendered-apps-such-as-minecraft-remotely-0-2
I have tested this and it seemed to work fine for me.

closed 2021-02-28 by FD-