FD-/RPiPlay #222: No signal detected by the monitor

mijalapenos, 2021-02-18:
Hello, I am trying to setup RPiPlay and from what I can tell, everything is running fine. Code compiled without errors and the RPiPlay is receiving packets when I connect to it (based on the messages in debugging mode). However, the monitor does not detect any signal in the mirroring mode. I ran it with command: `rpiplay -n "Apple TV" -b on -a hdmi -d`.

I assume it is something with the Lite Raspbian. Is the full desktop version required or is someone running it on the Lite version?

I am using Raspberry Pi Zero W and a 2014 MacBook Air with Big Sur installed.

FD-, 2021-02-18:
I'm using Raspbian Lite as well, so that's not the issue.

mijalapenos, 2021-02-18:
Thank you for the reply. I wanted to reply with the log to have it checked but it works now for some reason. Sorry for the false alarm.

closed 2021-02-18 by mijalapenos